boltgolt/howdy #1: Error after running learn.py

akabhirav, 2018-02-01:
> Traceback (most recent call last):
>   File "learn.py", line 82, in <module>
>     captureFrame(delay)
>   File "learn.py", line 27, in captureFrame
>     ref = face_recognition.load_image_file(tmp_file)
>   File "/home/akxer/.local/lib/python3.6/site-packages/face_recognition/api.py", line 81, in load_image_file
>     return scipy.misc.imread(file, mode=mode)
>   File "/home/akxer/.local/lib/python3.6/site-packages/numpy/lib/utils.py", line 101, in newfunc
>     return func(*args, **kwds)
>   File "/home/akxer/.local/lib/python3.6/site-packages/scipy/misc/pilutil.py", line 164, in imread
>     im = Image.open(name)
>   File "/home/akxer/.local/lib/python3.6/site-packages/PIL/Image.py", line 2543, in open
>     fp = builtins.open(filename, "rb")
> FileNotFoundError: [Errno 2] No such file or directory: '/tmp/howdy_user.jpg'

Boltgolt, 2018-02-01:
This sounds like you're using an invalid `/dev/videoX` device, but i would need more data. I've updated the learn.py file, maybe this will be more clear on what's going wrong?

If you're able, could you be so kind to give me the output of the following commands:
```
sudo ls /dev/ | grep video
```
```
sudo lsusb -vvvv | grep iFunction
```

akabhirav, 2018-02-12:
Sorry for the late reply

Output of `sudo ls /dev/ | grep video`

`video0`

Output of `sudo lsusb -vvvv | grep iFunction`

`iFunction               5 Integrated Webcam`

Boltgolt, 2018-02-12:
Looks like you only have 1 webcam, are you sure it's an IR camera? 

Either way, you need to change the `device_id` to 0 in `config.ini`. Be warned that this could lead to using the normal visible light camera which is a lot less secure.

Boltgolt, 2018-03-01:
As this issue was opened when Howdy was quite different than it is now, this issue has gone stale. I'm closing this issue but please do open a new one if you still experience problems with the current Howdy.

closed 2018-03-01 by Boltgolt