boltgolt/howdy #2: Error when running sudo command

Leviathan220, 2018-02-09:
$ sudo -i
Traceback (most recent call last):
  File "/home/user/howdy/compair.py", line 5, in <module>
    import face_recognition
ModuleNotFoundError: No module named 'face_recognition'
Unknown error: 1

Leviathan220, 2018-02-09:
Also I installed everything correctly and I can import facial_recognition into both python and python3
Also running
$ python3 learn.py
Completes successfully

Boltgolt, 2018-02-09:
Could you try running the compare ("compair") script directly with `python3 /home/USER/howdy/compair.py`? It might still have PATH issues with the pam-python container.

Leviathan220, 2018-02-09:
$ python3 /home/USER/howdy/compair.py
$
I believe it runs it doesn't have an error just brings me back to the shell promp

Leviathan220, 2018-02-09:
sorry missclick still having trouble

Leviathan220, 2018-02-09:
Well odd but i left my laptop sitting on the coffee table walked outside to smoke a cigarette walked back inside and boom it logged me in.
so naturally i got excited opened a terminal
$ sudo -i
Traceback (most recent call last):
  File "/home/USER/howdy/compair.py", line 5, in <module>
    import face_recognition
ModuleNotFoundError: No module named 'face_recognition'
Unknown error: 1
I don't understand how magically it works then breaks again

Boltgolt, 2018-02-10:
How... magical

Howdy should only try to log you in until it hits a timeout of about 4 seconds (depending on the machine). That way it will fall back to password authentication if howdy isn't working. This means it should have stopped scanning _way_ before you got back to your laptop.

Did the IR emitters light up when you got back? Does it work when manually locking the machine (`super (windows) key` + `L`)?

Leviathan220, 2018-02-11:
You would be correct my laptop was opened but locked when I came back to it. As for (super (windows) key + L) well interestingly yes it does. It also works every time I lock my laptop even some odd times i thought for sure I was too far away good job lol. This does not work when I turn my laptop on however. Also sudo authentication does not work at all.

Leviathan220, 2018-02-11:
I just don't understand how the same piece of code can fail to import a library for a sudo request but succeed flawlessly for a login. Is it using different installed versions of python that are magically deleting themselves when I log in. I mean that is a joke obviously I do not believe that is the case. Still confusing though

Boltgolt, 2018-02-11:
It's very magical, as the exact same line activates the script for _any_ authentication request.

Could you try running

```
whereis python3
```

This should give you an absolute path to the python executable. In my case this is `/usr/bin/python3.5`.
In pam.py, on line 19, replace `"python3"` with that full absolute path (`"/usr/bin/python3.5"`).

Could you also check the output of 

```
pip3 show face_recognition
```

Which should give you the location where the face_recognition module was installed.

Leviathan220, 2018-02-11:
Running
`$ whereis python3
python3: /usr/bin/python3.6 /usr/bin/python3.6m-config /usr/bin/python3 /usr/bin/python3.6-config /usr/bin/python3.6m /usr/lib/python3.6 /usr/lib/python3.7 /usr/lib/python3 /etc/python3.6 /etc/python3 /usr/local/lib/python3.6 /usr/include/python3.6 /usr/include/python3.6m /usr/share/python3 /usr/share/man/man1/python3.1.gz`

Also running
`$ pip3 show face_recognition
Name: face-recognition
Version: 1.2.1
Summary: Recognize faces from Python or from the command line
Home-page: https://github.com/ageitgey/face_recognition
Author: Adam Geitgey
Author-email: ageitgey@gmail.com
License: MIT license
Location: /home/USER/.local/lib/python3.6/site-packages
Requires: Pillow, dlib, scipy, numpy, face-recognition-models, Click`

Leviathan220, 2018-02-11:
I am not sure if facial_recognition is supposed to be installed into my home directory but that is where pip installed it

Leviathan220, 2018-02-12:
I did not use sudo when installing it nor did I run it in any kind of virtual environment

Leviathan220, 2018-02-12:
It is litterally reinstalling it to the same directory is there a way to specify it for the right directory

Boltgolt, 2018-02-12:
Try running it _with_ sudo, i don't think that will ever install to a .local.

Leviathan220, 2018-02-12:
I'll be damned that worked lol. Thanks so much for the help I really appreciate it.
P.S. keep making cool little things like howdy

Leviathan220, 2018-02-12:
I mean using sudo -i is really slow
slower than typing in my password.
but logging in is quick?

closed 2018-02-12 by Leviathan220

Boltgolt, 2018-02-12:
Great to hear that it's working! I'm close to finishing an installer on the dev branch so issues like this can't happen anymore.

The sudo command being slower could be a whole host of other issues. It might be CPU limited or not have the right models. Deleting the `models` folder and running `python3 learn.py` again will retrain it.

Leviathan220, 2018-02-13:
That might be something I will try since I have moved the folder into the pam.d directory and made it read only. I am impressed however that the camera on my inspiron-5379 is so good at capturing my face in low light like no light at all. Maybe it is just the deep learning or an error in setting up the error rate. I left that at default for the scanning like what is the difference in 3 vs 1-4. I know the readme says not to use anything above 5. I want the most security obviously. Having this integrated into gnome shell in future would be amazing. Maybe something like the cpu-power-manager [https://github.com/martin31821/cpupower](url) where you install the extension and then it from there asks permission and downloads all the needed libraries and compiles them. It would also make for easy cleanup if you ever wanted to remove it.

Leviathan220, 2018-02-13:
though it could also be slow because I am using zsh instead of bash or like you said a whole host of other issues. However it has been a bit quicker recently maybe moving the directory who knows lol.

Boltgolt, 2018-02-13:
- If you have the type of camera i have, it has IR emitters that light up your face in infrared light. This means it's completely independent of ambient light and can even work in pitch blackness. 
- The algorithm gives the face recognition a reliability score, where 0 is without a doubt. I score under three most of the time. Try setting it to 1, it should fail every time.
- I've been working on an installer just like that. If you take a look at the readme in the [dev branch](https://github.com/Boltgolt/howdy/tree/dev) it is a lot simpler than it is now.
- On-disk encryption makes no sense without a user password. The issue is that if the program can read a private key, so can the attacker. (This is also why your saved passwords in chrome are not encrypted) I think the best way to secure it is to keep the permissions as strict as possible. 
- Even importing the face recognition library makes 1 of my 8 CPUs spike to 100%, and during recognition all of them are at max. It's really intense to run. I don't think it has anything to do with zsh/ bash though, as PAM is completely independent of those.

Boltgolt, 2018-02-14:
I just pushed a new version that's almost a complete rewrite, it might fix the slowness a little and would have prevented the pip sudo issue.

Leviathan220, 2018-02-14:
Should I uninstall all the pip libraries I installed or would simply deleting the git repo be enough

Boltgolt, 2018-02-14:
Deleting it should be enough, but don't forget to also remove it from the common-auth file in pam.d

Leviathan220, 2018-02-14:
Ah good call. Idk why all of your cores go to 100%. Only 1 of mine gets pegged for like a second regardless of if I am importing or using it for authentication.