boltgolt/howdy #5: Certainty

Leviathan220, 2018-02-14:
For some reason distance is a huge struggle now. I have taken 6 scans so far and using my pc at the same distance as before it fails to detect a face. I also noticed you changed the certainty to 3.5 instead of 3. This should make it less accurate and more likely to authenticate. Only the opposite is happening. I can fix this by getting really close to my IR camera but it is inconvenient to do and I did not have to prior.

Leviathan220, 2018-02-14:
Basically the program went from working almost 100% of the time to <10%.

Leviathan220, 2018-02-14:
I also tried removing and remaking models and using just 1 model

Leviathan220, 2018-02-14:
Why did I update lol. Like the number 1 rule if it isn't broke don't fix it.

Boltgolt, 2018-02-14:
I've added 2 new important config values. `end_report` gives a detailed report after a succeeded scan, including the used resolution and winning certainty. `max_height` scales down the frames from your webcam. This gives a dramatically higher fps, but can make scanning harder.

You can find the config in `/lib/security/howdy/config.ini`. Try setting it to a higher max and upping the certainty to 6.

Leviathan220, 2018-02-14:
changing the `max_height` does not seem to anything as the output of adding a face is the same `Adjusting resolution from 384x288 to 340x374.`

Leviathan220, 2018-02-14:
In fact changing any value located in the config `/lib/security/howdy/config.ini` seems to have no effect on the program.

Boltgolt, 2018-02-14:
That's not where the config makes the change though, the add command will always capture in full resolution as it makes the models more reliable. You could try renaming the config.ini to something else and look if the program still runs. I'd recommend not adding more than 3 models as every additional model slows down the script a lot.

I think your models are just not accurate enough in some way. The actual capture and compare code hasn't changed a lot between these 2 releases.

Boltgolt, 2018-02-14:
If it's having so much trouble finding you in the add command it will have even greater issues when trying to authenticate you. I'll look into a test function though, so you can see more clearly what's going wrong. It could be that you're just not illuminated enough.

Leviathan220, 2018-02-14:
But it was working perfectly and I haven't changed anything other than update Ubuntu and download the newest version of howdy

Leviathan220, 2018-02-14:
Like if it just never worked then ok sure maybe it's just buggy on my hardware or something. Though I used it flawlessly for about a day or two. It was a bit slow at first but that may have been my imagination cause it was definitely quicker and more convenient than typing my password everytime after that. It also had a higher certainty value of 3

Leviathan220, 2018-02-14:
I am gonna uninstall everything and try again. I think honestly it is the pip programs being screwy again. Also if the new version of howdy installs them automatically it is probably best i uninstall anyway.

Leviathan220, 2018-02-14:
I tried uninstalling all the libraries manually it didn't help.

Boltgolt, 2018-02-15:
I think the add command (the former `learn.py`) needs a serious rework. In the meantime i've pushed a new `test` command to the dev branch. Could you try it and see if it can detect your face?

To download the lastest dev commit:
```
sudo -i
cd /lib/security/howdy/
git fetch
git checkout dev
pull
```

You should now have a test command which you can run anywhere with `sudo howdy test`.

Leviathan220, 2018-02-16:
Whelp.
`Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

No protocol specified
Unable to init server: Could not connect: Connection refused

(Howdy Test:2083): Gtk-WARNING **: cannot open display: :0
`

Leviathan220, 2018-02-16:
Nvm that is an issue with ubuntu using wayland on 17.10. Yes it detects my face.

Boltgolt, 2018-02-16:
If the test command can see you it must be a timing issue with your emitters. I've pushed yet another update to dev with a revamped add command, try pulling it and see if it works better.

Leviathan220, 2018-02-18:
Congratulations I can scan my face. However authentication is super slow. Like upwards of almost 10 seconds sometimes before the ir camera even starts flashing.

Leviathan220, 2018-02-18:
Nvm the slowness was my cpu governor being set to 30 percent. I forget how much horsepower that this deep learning stuff takes.

Leviathan220, 2018-02-18:
Nope I spoke too soon
`sudo -i
Face detection timeout reached`

Boltgolt, 2018-02-20:
I can't really do much about that, if the `test` command shows it's seeing a face then it's just not recognizing you. You could try upping the certainty and adding more models. I'd recommend moving your current models folder to a safe location (just in case).

Leviathan220, 2018-02-21:
Well where would you move them? Also the program is mostly working now. Sometimes it fails but unfortunately I have not had much time to play around with it. I am gonna close this issue though. While the program is definitely buggy it is working.

closed 2018-02-21 by Leviathan220

Boltgolt, 2018-02-21:
Anywhere is fine, it's just a backup. This script could possibly be much more reliable with an algorithm that's specifically made for black and white IR pictures. It would take ages to train that tough and it has so few users that it's just not worth it at the moment.

Leviathan220, 2018-02-23:
Well I would be more than willing to provide my hardware for testing and training this if it get's me closer to a better login. However it would be really nice if the I could get more feedback from failed attempts. Such as the score I got compared to what I need to achieve successful authentication. It may also be helpful to save those images for later tweaking of the algorithm. I mean with everyone and their brother adding a connect into everything from phones to laptops. The actual user base may limited at the moment. However the technology is very powerful. Like imagine this linked with something such as OpenPGP smartcards for two factor authentication. Now this is assuming Howdy works properly as intended. Though I still have fun playing around with it.

Leviathan220, 2018-02-23:
Maybe one day. As matching IR imagery and the visible spectrum images together is probably one of the hardest things for AI and DNN right now.

Boltgolt, 2018-02-23:
The issue is that we would need a _lot_ of training images. Normal photos in the visible spectrum are easy to find, but IR photos are really different.

Leviathan220, 2018-03-01:
Yes but finding people for training images becomes a lot easier once you have tools available to do so.

Boltgolt, 2018-03-01:
What do you propose as tools?