boltgolt/howdy #6: IR not recognized

nico151999, 2018-02-15:
It's a really awesome project and face recognition worked great before your merged the dev branch into master. Most probably it still does now. I haven't tried it with the new version yet as the program still does not recognize the IR one but my normal webcam only (HP Spectre 15" x360 Coffee Lake). I took a look inside /dev and the only videoX device that's found is video0 which is my webcam. I have no idea which of the other devices the IR sensors could be. Thus I've also no idea how to extend the repo to support my device. Maybe you can filter the right device from the following "ls -1 /dev"-output:
```
acpi_thermal_rel
autofs
block
btrfs-control
bus
char
console
core
cpu
cpu_dma_latency
cuse
disk
dri
drm_dp_aux0
drm_dp_aux1
drm_dp_aux2
ecryptfs
fb0
fd
full
fuse
gpiochip0
hidraw0
HID-SENSOR-2000e1.11.auto
HID-SENSOR-2000e1.12.auto
HID-SENSOR-2000e1.14.auto
HID-SENSOR-2000e1.17.auto
HID-SENSOR-2000e1.1.auto
HID-SENSOR-2000e1.20.auto
HID-SENSOR-2000e1.21.auto
HID-SENSOR-2000e1.24.auto
HID-SENSOR-2000e1.25.auto
HID-SENSOR-2000e1.26.auto
HID-SENSOR-2000e1.27.auto
HID-SENSOR-2000e1.2.auto
HID-SENSOR-2000e1.3.auto
HID-SENSOR-2000e1.4.auto
HID-SENSOR-2000e1.5.auto
HID-SENSOR-2000e1.6.auto
HID-SENSOR-2000e1.8.auto
hpet
hugepages
hwrng
i2c-0
i2c-1
i2c-2
i2c-3
i2c-4
i2c-5
i2c-6
iio:device0
iio:device1
iio:device2
iio:device3
iio:device4
iio:device5
initctl
input
kmsg
kvm
lightnvm
log
loop0
loop1
loop2
loop3
loop4
loop5
loop6
loop7
loop-control
mapper
mcelog
media0
mei0
mem
memory_bandwidth
mqueue
net
network_latency
network_throughput
null
nvme0
nvme0n1
nvme0n1p1
nvme0n1p2
nvme0n1p3
port
ppp
psaux
ptmx
pts
random
rfkill
rtc
rtc0
shm
snapshot
snd
stderr
stdin
stdout
tpm0
tpmrm0
tty
tty0
tty1
tty10
tty11
tty12
tty13
tty14
tty15
tty16
tty17
tty18
tty19
tty2
tty20
tty21
tty22
tty23
tty24
tty25
tty26
tty27
tty28
tty29
tty3
tty30
tty31
tty32
tty33
tty34
tty35
tty36
tty37
tty38
tty39
tty4
tty40
tty41
tty42
tty43
tty44
tty45
tty46
tty47
tty48
tty49
tty5
tty50
tty51
tty52
tty53
tty54
tty55
tty56
tty57
tty58
tty59
tty6
tty60
tty61
tty62
tty63
tty7
tty8
tty9
ttyprintk
ttyS0
ttyS1
ttyS10
ttyS11
ttyS12
ttyS13
ttyS14
ttyS15
ttyS16
ttyS17
ttyS18
ttyS19
ttyS2
ttyS20
ttyS21
ttyS22
ttyS23
ttyS24
ttyS25
ttyS26
ttyS27
ttyS28
ttyS29
ttyS3
ttyS30
ttyS31
ttyS4
ttyS5
ttyS6
ttyS7
ttyS8
ttyS9
uhid
uinput
urandom
userio
v4l
vcs
vcs1
vcs2
vcs3
vcs4
vcs5
vcs6
vcsa
vcsa1
vcsa2
vcsa3
vcsa4
vcsa5
vcsa6
vfio
vga_arbiter
vhci
vhost-net
vhost-vsock
video0
zero
```

Any support is greatly appreciated. Count on my support in case I can help. Thanks in advance!

nico151999, 2018-02-15:
All right, so the face recognition still works. However, I guess I have even less experience with the media API of the Linux Kernel than you have, I've never touched it before. I'd not even know where to start implementing the IR emitters. Really disappointing...

Boltgolt, 2018-02-16:
I've looked into the mechanics of it all, but with my limited knowledge of C++ it's near impossible. If you really want to, this [TI](http://processors.wiki.ti.com/index.php/UserGuideOmap35xCaptureDriver_PSP_04.02.00.07) site helped me understand the system a lot better, but i can't even figure out where i'm supposed to import the header file from.

Boltgolt, 2018-03-31:
@nico151999 @paulobrien @bhanu6599

I've come across an option in the `v4l2-ctl` command that can list all available controls for that camera. Could you try running

```
ls /dev/ | grep video
```

and then running the following command for each video device it reports

```
v4l2-ctl -l -d /dev/videoX
```

There might be a related control option for IR in there somewhere.

nico151999, 2018-03-31:
Thanks for coming back to this issue @Boltgolt 
It's interesting to see what options the device offers. It seems like there's no IR option, however.
```
nico@NicosSpectre:~$ v4l2-ctl -l -d /dev/video0
                     brightness (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast (int)    : min=0 max=255 step=1 default=32 value=32
                     saturation (int)    : min=0 max=100 step=1 default=64 value=64
                            hue (int)    : min=-180 max=180 step=1 default=0 value=0
 white_balance_temperature_auto (bool)   : default=1 value=1
                          gamma (int)    : min=90 max=150 step=1 default=120 value=120
                           gain (int)    : min=1 max=7 step=1 default=1 value=1
           power_line_frequency (menu)   : min=0 max=2 default=2 value=2
      white_balance_temperature (int)    : min=2800 max=6500 step=1 default=4000 value=4000 flags=inactive
                      sharpness (int)    : min=0 max=7 step=1 default=2 value=2
         backlight_compensation (int)    : min=0 max=1 step=1 default=0 value=0
                  exposure_auto (menu)   : min=0 max=3 default=3 value=3
              exposure_absolute (int)    : min=10 max=626 step=1 default=156 value=156 flags=inactive
         exposure_auto_priority (bool)   : default=0 value=1
```
I simply guess there are two video devices: the IR one and the usual webcam. Both need an own driver. The webcam has a linux driver whereas the IR cam doesn't have one. Disappointing...

bhanu6599, 2018-04-01:
Only video0 is visible. My ouput has the same parameters as that of @nico151999 .

techied, 2018-06-03:
I have the same issue here. I have some experience in C however, so I'm willing to give it a shot. I'll look at those links.
Output of v4l2-ctl is the same as above:
`                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=0 value=0
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=1 max=8 step=1 default=1 value=1
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=10 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=3 value=3
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
              exposure_absolute 0x009a0902 (int)    : min=10 max=625 step=1 default=166 value=625 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0

`

minzukas, 2019-07-15:
Same here. Face recognition works, though no IR emitters at all (works well in windows).

[image]