boltgolt/howdy #7: Not working

szelag82, 2018-02-20:
hi, I receive the following errors and the howdy command is not available after installation (ubuntu 17.10, IR led blinks during the test):

Traceback (most recent call last):
  File "/tmp/howdy_install.py", line 94, in <module>
    handleStatus(subprocess.call(["git", "clone", "https://github.com/davisking/dlib.git", "/tmp/dlib_clone"]))
  File "/usr/lib/python3.6/subprocess.py", line 267, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.6/subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1344, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'git': 'git'

Boltgolt, 2018-02-20:
Missed a dependency, sorry. The merge i just did should fix this.

szelag82, 2018-02-20:
those errors have disappeared but I have a new one:

Traceback (most recent call last):
  File "setup.py", line 40, in <module>
    from setuptools import setup, Extension
ModuleNotFoundError: No module named 'setuptools'

Boltgolt, 2018-02-20:
Looks like you don't have pip either, pushed another one to fix that. I'll be here if you get another error.

szelag82, 2018-02-20:
new message at the time of installation (but this is not a error):

The directory '/home/szelag/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/szelag/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already up-to-date: pip in /usr/lib/python2.7/dist-packages

Nothing has changed at the end:

Traceback (most recent call last):
  File "setup.py", line 40, in <module>
    from setuptools import setup, Extension
ModuleNotFoundError: No module named 'setuptools'

Boltgolt, 2018-02-20:
Missed one dependency the last time i added packages, that one should work now.

szelag82, 2018-02-20:
nope, still the same error:

Traceback (most recent call last):
  File "setup.py", line 40, in <module>
    from setuptools import setup, Extension
ModuleNotFoundError: No module named 'setuptools'
Error while running last command

Boltgolt, 2018-02-20:
Fixed setuptools only being available on python2, hopefully you should get past that now?

szelag82, 2018-02-20:
new error :-)

Invoking CMake setup: 'cmake /tmp/dlib_clone/tools/python -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/dlib_clone/build/lib.linux-x86_64-3.6 -DPYTHON_EXECUTABLE=/usr/bin/python3 -DUSE_AVX_INSTRUCTIONS=yes -DDLIB_USE_CUDA=no -DCMAKE_BUILD_TYPE=Release'
error: [Errno 2] No such file or directory: 'cmake': 'cmake'
Error while running last command

I have to go to sleep, I'll be back tomorrow to test. thank you for your time!

szelag82, 2018-02-20:
sorry, I installed cmake and the installation went on, but there is still an error:

  File "/usr/lib/python3.6/subprocess.py", line 291, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
Error while running last command

Boltgolt, 2018-02-21:
That looks nasty. What version of Ubuntu are you running and has nothing else been printed to the console?

magarto, 2018-02-25:
I have the same error like szelag82

> File "/usr/lib/python3.6/subprocess.py", line 291, in check_call
> raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
> Error while running last command

Full error is:

/tmp/dlib_clone/dlib/external/pybind11/include/pybind11/detail/common.h:111:10: fatal error: Python.h: No existe el archivo o el directorio
 #include <Python.h>
          ^~~~~~~~~~
compilation terminated.
CMakeFiles/dlib_python.dir/build.make:134: recipe for target 'CMakeFiles/dlib_python.dir/src/svm_c_trainer.cpp.o' failed
make[2]: *** [CMakeFiles/dlib_python.dir/src/svm_c_trainer.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/dlib_python.dir/all' failed
make[1]: *** [CMakeFiles/dlib_python.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Traceback (most recent call last):
  File "setup.py", line 246, in <module>
    'Topic :: Software Development',
  File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 129, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 170, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 156, in call_command
    self.run_command(cmdname)
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/install_lib.py", line 24, in run
    self.build()
  File "/usr/lib/python3.6/distutils/command/install_lib.py", line 109, in build
    self.run_command('build_ext')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 126, in run
    self.build_extension(ext)
  File "setup.py", line 162, in build_extension
    subprocess.check_call(cmake_build, cwd=build_folder)
  File "/usr/lib/python3.6/subprocess.py", line 291, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
Error while running last command

magarto, 2018-02-25:
Update: Real error is
/tmp/dlib_clone/dlib/external/pybind11/include/pybind11/detail/common.h:111:10: fatal error: **Python.h: File doesn't exist**

magarto, 2018-02-25:
No, I could install everything but "sudo -i" shows the following error:
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 11, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
Unknown error: 1

"pip3 install openv-python" doesn't fix it. Also "python3" and "import cv2" commands work, so maybe pam is calling to python2 instead of python3.

Boltgolt, 2018-02-25:
Thanks so much for figuring this out, i'll fix the python2 apt packages. 

I can't reproduce the last 2 errors, and i can't really expect everyone to have conda installed. Might need to add information on how to fix that installation error in the readme.

I've added the last note as a new issue at #9. However, i'm not sure how this can be implemented. It needs to be done in PAM but i don't know enough about it to fix it and online manuals are few and far between.

Boltgolt, 2018-03-16:
As i've had no other reports of the specific issue with cv2, i'm going to close this for now. Feel free to comment if you've also encountered this error.

closed 2018-03-16 by Boltgolt

reverland, 2018-07-31:
apt install python3-opencv just works for me.
on ubuntu 18.04

boltgolt, 2018-07-31:
It also works as expected on the travis 16.04 instance, it seems very machine dependent

reverland, 2018-08-02:
maybe I should comment issue 32 but comment here...