boltgolt/howdy #9: Allow normal password entry at the same time as face recognition

Boltgolt, 2018-02-25:
As suggested by @magarto, make Howdy asynchronous so a password can be entered at the same time though the default Unix way. This has to be done with the PAM, not sure how.

Boltgolt, 2018-02-25:
I've found a [doc](http://www.ullrich-online.cc/fingerprint/doc/Hacking.html#idp36356224) from the fingerprint-gui module that explains how they got this principle to work.

magarto, 2018-02-26:
If I unterstand well from case 4, typing password should stop face_detection module (if not other modules are working).

> Case 4 (user has typed the password)
> If the PAM module returns from its password prompt with a non empty password field the user has typed his password at the prompt. Then the module sends SIGUSR2 to the helper to stop it immediately and returns PAM_IGNORE. So the other PAM modules can check the given username/password.

Boltgolt, 2018-02-26:
That's what I tried to do in python, but I ran into 2 issues: 

First of all, the call to the PAM conversation is interpreter-blocking. This means that when we promt for a password nothing else can run. Even separate threads will freeze. 

Secondly, I can't get the PAM_IGNORE signal to pass the captured password to the other PAM modules. It's doing the exact same thing as fingerprint-gui (I've run though the source) but it just falls through to a second password prompt.

RussianNeuroMancer, 2018-10-28:
@boltgolt did you tried contact linux-pam developers about this issue?

boltgolt, 2018-10-28:
I haven't, and looking at the [issues page](https://sourceforge.net/p/pam-python/tickets/) of the project it doesn't seem very active. I was working on removing the dependency in it's entirety and writing the PAM part in C++. However, i'm not that comfortable in C++ so that's on hold for now.

RussianNeuroMancer, 2018-10-29:
>  it doesn't seem very active

Indeed: https://sourceforge.net/p/pam-python/code/ci/19f932b71918cbcd639a26bef79e2de0c3d9d5c4/log/?path=

Maybe try this one?

https://github.com/minrk/pamela

dmig, 2018-11-19:
sourceforge project seems dead, but here is some activity: https://github.com/Ralnoc/pam-python

papermache, 2019-11-17:
LM 16.2 Tina has a virtual keyboard at login that accepts a password if camera is active.

papermache, 2019-11-18:
LM Tina 19.2 Tina has an onscreen keyboard at login however does not accept the PW when the camera is active.

icorbrey, 2020-09-28:
@boltgolt With your hesitation to use C++ maybe we could try using a different low-end language? There are several alternatives that would perform just as fast (if not faster) than plain old C++ (specifically Rust and Go come to mind).

Chandan-CV, 2021-09-19:
It's working for me!!
I can enter login the normal way and howdy works asynchronously.

Slackadays, 2023-03-08:
Any progress on this? I thought Howdy was always asynchronous like this, but if it isn't, then I don't know if it's compatible with my (or @papermache's) workflow.

boltgolt, 2023-03-09:
You could try the beta right now and see if it works as you would expect

musikid, 2023-10-13:
Closed by #484.

closed 2023-10-13 by musikid