boltgolt/howdy #13: How to show desktop automatically when PAM success on Ubuntu?

codingwaysarg, 2018-03-13:
Hi! I'm using Ubuntu and when I boot the PAM process on login is working great but I have to press login button after authentication. 
It's possible to go to desktop automatically? I think that it's not library 
responsibility, but how could I modify it to do that?

Thanks.

codingwaysarg, 2018-03-13:
If you could give me a hint, I'll gladly submit a pull request

Boltgolt, 2018-03-13:
Sorry for not replying, i was looking for a semi-reliable way to dismiss the lockscreen. The branch i just merged should add this if you flip the right config option to `true`.

codingwaysarg, 2018-03-13:
Hi! Thanks for fast reply. I've tested the code with variable set to true. It's not working for me, more precisely I'm using Ubuntu 17.10 with gdm.

Thanks.

Boltgolt, 2018-03-13:
What happens if you run `sudo loginctl lock-sessions --no-ask-password && sleep 10 && sudo timeout 1 loginctl unlock-sessions --no-ask-password`? For me this locks the screen, runs howdy and dismisses the lockscreen after 10 seconds.

codingwaysarg, 2018-03-13:
If I run it in a new terminal it simply run howdy (because of sudo in the command), identify my face and then a fast blink in the screen happens, nothing else. If I run the command again, then it don't run howdy (because already identified), lock the screen and after 10 seconds unlock the screen.

Boltgolt, 2018-03-13:
That's weird, that it does actually work when using the command in a shell. Are you 100% sure that it is enabled in the right config file?

codingwaysarg, 2018-03-13:
Sorry, your command is working ok, the fast blink is because already enabled dismiss option on config, so right after `sudo loginctl lock-sessions --no-ask-password` and howdy identifies me, `sudo timeout 1 loginctl unlock-sessions --no-ask-password` executes.

It's weird that it's not working when I reboot the pc, maybe longer timeout?

Boltgolt, 2018-03-13:
I'm not 100% sure why it fails sometimes, and i can't reproduce the fail conditions after i have one. It's a bit temperamental. I've marked it as experimental in the config and it should be seen as such for now.

If you do see some pattern in the failures, please let me know.

PR #14 (2018-03-13): Added automatic lockscreen dismissal config option
This fixes #13 and adds a new config option to dismiss the lockscreen immediately after we recognize the user.
closed 2018-03-13 by Boltgolt