boltgolt/howdy #17: Spreading the word

Leviathan220, 2018-04-01:
Wasn't sure how to email you. I made a post about the project on the Level1 forums [https://forum.level1techs.com/t/windows-hello-style-authentication-for-ubuntu/126171](url). if you don't already know who they are you should totally go check them out they are great. Also if you wanted to add anything to my post feel free. I tried to make a joke but up to you if you find the humor in it.

Boltgolt, 2018-04-01:
Thanks so much for taking the time to do this! Reading your forum post has honestly made my day. Howdy was originally intended as a very simple script for just me to use, but seeing it getting installed in the wild is amazing. Again, thanks for hanging around!

closed 2018-04-01 by Boltgolt

Leviathan220, 2018-04-02:
Hey software is all about the community especially in open source. I mean I have really enjoyed playing around with this for various authentication types. There are plenty of people who I could see wanting something like this but maybe don't have all the know how to do it themselves.