boltgolt/howdy #19: IR emitter don't blink with cv2.VideoCapture

janecz-n, 2018-04-07:
I juste receive my new computer with an IR Camera so I would like to try your software ! :)
So I have properly done the installation, selection the device with blinking IR emitter.
But when I `sudo` or use howdy test my emitter blink only once.

I have try to manually:
```
import cv2
vd = cv2.VideoCapture(int("0"))
vd.read()
```
And like in howdy test, emitters blink once and then stay in a 'low red state' and don't blink anymore even if I vd.read() again.

Any idea to where I should look ?

janecz-n, 2018-04-07:
The emitters light up at low intensity and don't blink.
For now the only command who make them blink is `fswebcam -S 9999999999 -d /dev/video0`

Boltgolt, 2018-04-07:
That sounds like a driver issue. Howdy depends on linux webcam drivers supporting the IR cameras and in a lot of cases the driver just doesn't work. It is weird though that fswebcam can make them blink while skipping frames, but i'm not sure how to bring that to howdy if cheese can't even do that.

Teifun2, 2018-10-03:
I would like to mention that i have the same Problem with the Thinkpad T480.
What i find strange is taht during the test in the installation (Ubuntu through apt) the IR Emitters work as they should. Flickering on high power. How can that be?

boltgolt, 2018-10-04:
Somehow the emitters seem to work correctly on certain frame sizes, see #51

Teifun2, 2018-10-04:
Oh cool. Yes this worked for me as well with 320x320 (and other small resolutions). It seems tough as if the Frame Grabber and the IR Emitters are out of sync because it can happen (in the test) that sometimes there is no "scan frame" for more than one second. I assume this is a Problem with the driver.  But it works  thanks allot.