boltgolt/howdy #22: Face detection timeout reached

klascano, 2018-04-19:
hi, 

Great work, but, i have to add new face profile often, i know what happen. if you need information i would like to help.

klascano, 2018-04-19:
Face detection timeout reached

Boltgolt, 2018-04-19:
Have you ever had an successful howdy authentication?

klascano, 2018-04-19:
yes, but each day i have to update my face profile to work.

Boltgolt, 2018-04-21:
Your camera might not be good enough for realiable face detection, could you try to set the `certainty` to 5 with `sudo howdy config`?

Boltgolt, 2018-04-30:
Any progress on this one?

klascano, 2018-05-03:
Hi!

yes, i changed certainty = 5 and i work great, however, i noticed that when you dont watch directly to the camera and the light is poor its fails the login.

One important issue that i have is when a gui app require root permission and the face detection fails, the root pass dialog doesnt appear so the app freeze for ever for example when you install virtualbox additions the main app wait ...........

Thanks, Great Job!

Boltgolt, 2018-05-03:
Good to hear. GUI prompts _should_ fall back to a password field, does `gksudo ls` work after a timeout?

closed 2018-05-03 by Boltgolt

klascano, 2018-05-04:
nope, it doesnt work, no appear password promnt dialog. terminal wait forever.
[image]

Boltgolt, 2018-05-04:
I think this might be a duplicate of #29, are you using an IR camera?