boltgolt/howdy #26: Support for Fedora

KelleyMcChes, 2018-04-25:
I don't know if this is a lot of work or not, but would it be possible for Howdy to be ported to Fedora? If that is too much work, could you add documentation so that I could compile from source?

Boltgolt, 2018-04-25:
I'd love to have a RPM build of Howdy, but i have close to 0 experience with fedora and related distros. I don't have a clue how to build or distribute RPMs either.

There also might be fundamental incompatibilities between the systems, `face-recognition` seems to work on Fedora but installing `dlib` might be harder. The `libpython-pam` module is also critical, 

You could start by trying to get the [very first version](https://github.com/Boltgolt/howdy/tree/4b37c5e813e65e79eb10336436ddd804b8865e2b) to work. This version didn't have an installer yet and instead has a lengthy installation guide. It all depends how much effort you want to put into this.

Boltgolt, 2018-05-01:
Any progress by any chance?

bar-code, 2018-08-07:
thanks to @boltgolt I shall working on RPM version of howdy and if it's complete I shall notice it here !

mrepetto94, 2018-09-06:
Any news about the Fedora RPM package? @bar-code

boltgolt, 2018-10-04:
No visible progress on the fork of @bar-code, marking this as stale again. Add a comment if your up for making a port for Fedora.

luyatshimbalanga, 2018-11-26:
Hello @boltgolt ,
I am a Fedora package maintainer. Here is the initial built on [COPR repository](https://copr.fedorainfracloud.org/coprs/luya/howdy/). 
SPEC file: https://copr-be.cloud.fedoraproject.org/results/luya/howdy/fedora-29-x86_64/00829419-howdy/howdy.spec

Currently missing is the python-face_recognition_models not yet avaiable in the repository.

boltgolt, 2018-11-26:
Great work, thanks! The good news is that @sapjunior and @dmig have been working on a much much faster version that does not use face_recognition over at #99. 

What would Fedora users need to to install that package? Would you like to create a pull request on the dev branch with Fedora installation instructions in the readme?

dmig, 2018-11-26:
It would be better, if someone makes package build script, similar to `debian/`

boltgolt, 2018-11-26:
The Arch package also has its PKGBUILD on AUR and no packaging files in this repo. If it's an option for Arch it should also be allowed for Fedora. 

I'd much rather have both of them stored in git but not all packaging systems are the same in that regard.

dmig, 2018-11-26:
Let's leave that to someone who use these distros.

luyatshimbalanga, 2018-11-28:
Both face_recognition and its models are packaged in Copr repository. Howdy is on the latest stable release ready for testing.

luyatshimbalanga, 2018-11-28:
Next step will be writing a Fedora guideline installation subject to change once the package landed on the official repository.

boltgolt, 2018-11-29:
The packages on Copr look good, pull request #103 looks good as well. Closing after PR merge.

williamwlk, 2018-11-30:
Hi @luyatshimbalanga and @boltgolt 

Had a look at the Copr repo and tried to get the hang of it. 

python2 and python3 can co-exist among other things. (Check)
And I could be wrong but 'python3-face_recognition_models ' package is missing in action.
In addition, how about pam_python.so? 

python3-face_recognition_models could be a temporary non issue. But pam_python.so is nowhere to be found or addressed.

Thoughts?

boltgolt, 2018-11-30:
The [face_recognition_models package](https://copr.fedorainfracloud.org/coprs/luya/howdy/package/python-face_recognition_models/) seems included in the Copr repo. Isn't [pam-python](https://fedora.pkgs.org/28/rpm-sphere/pam_python-1.0.6-2.1.x86_64.rpm.html) available the official reposotory?

williamwlk, 2018-11-30:
Hi @boltgolt 

Yes. It is being built and packaged OK but it is not where it is supposed to be. Could be temporary and non issue.

pam_python is I am afraid not available at the official Fedora Repo.

That's why I have had to build it on my RHEL7.5. (STEP 5 (Prep Prep Prep for pam_python.so)).

Ref: 
"Pam-python is part of Debian. Most Debian derived distribution can install using apt-get."
http://pam-python.sourceforge.net/

williamwlk, 2018-11-30:
> pam_python is I am afraid not available at the official Fedora Repo.
> 

Ohh don't mind my oversight! It is at RPM Sphere Repo as you have pointed out.

pam_python-1.0.6-2.1.x86_64

So this is cool now.

williamwlk, 2018-11-30:
> Yes. It is being built and packaged OK but it is not where it is supposed to be. Could be temporary and non issue.

Got it installed OK.

> python3-face_recognition_models-0.3.0-1.fc29.noarch

This is cool too.

fswebcam-20140113-10.fc29.x86_64 is there too. So that you can do preinst script. and postinst too.

williamwlk, 2018-12-02:
Thank you. @luyatshimbalanga and @boltgolt 

How do you intend to help the end users configure the webcam in the config.ini?

pre/postinstall scripts?

luyatshimbalanga, 2018-12-03:
@williamwlk 
Looking at both debian pre/postintall scripts, they should be part of howdy in the event an user switched on another camera. I look at a webcam application about the initial autoconfiguration with minimum user interaction.

boltgolt, 2018-12-03:
To reconfigure the webcam a user needs to edit the config by hand to the right path. I wouldn't spend too much effort on it as that part will move to the UI in Howdy 2.6.0.

williamwlk, 2018-12-03:
@boltgolt No problem. And no worries at all. I was just curious as your debian pre/postinstall scripts were there and did all the stuff for the end users solid.

For most sys admins (of Fedora/CentOS/RHEL), they sure can find a way to do just that. 

To attach an install note would be a wise thing to do though.

And a good idea really re UI in Howdy 2.6.0.

Thank you for the beautiful software piece for the linux world.

boltgolt, 2019-01-06:
With Howdy being packaged on COPR with installation instructions in the readme i'll close this issue for now

closed 2019-01-06 by boltgolt