boltgolt/howdy #27: Pink image returned by test?

arifeinberg, 2018-04-25:
This may be outside the scope, but I'm having trouble adding a model with my IR cam (04f2:b613), and upon using test, I see that the camera is returning a mostly pink image:
[image]

I also notice that while it's running, the IR emitters are constantly on, not flashing.   I notice that when adobe flash uses the IR cam (for something else), the IR emitters flash, and a readable grayscale image is produced.  I have no idea where this is controlled.  Is there a parameter specified by the application that calls the camera, or is this determined somewhere else?

arifeinberg, 2018-04-25:
The Video4Linux preview also returns a usable image.  It seems this camera offers two modes: Motion-JPEG and YUYV raw.  Motion-JPEG is the one that works.  The mode is not set at the level of this software?

Boltgolt, 2018-04-25:
Could you try replacing line 22 (a white line) in `/lib/security/howdy/cli/test.py` with the following:

```
video_capture.read()
print(video_capture.get(cv2.CAP_PROP_FOURCC))
video_capture.set(cv2.CAP_PROP_FOURCC, 1196444237.0)
```

And see if that fixes anything or throws any errors? I'm not at my PC right now so it might be a little off.

arifeinberg, 2018-04-25:
I was just a few minutes behind you!  I added the following line after initialization of video_capture (in add, test, and compare):
video_capture.set(cv2.CAP_PROP_FOURCC, 1196444237)
and it is now working (after I raised the dark threshold).

Thanks!

Boltgolt, 2018-04-26:
Great! I'll add it as a setting in a later release

closed 2018-05-10 by Boltgolt