boltgolt/howdy #28: Module not being called in mate lockscreen

arifeinberg, 2018-04-26:
I'm running ubuntu-mate and I have howdy working normally when I sudo or login through lightdm, but when I lock my screen, I can only unlock with my password.  The webcam never turns on when I get back to the lockscreen.  My last computer (also ubuntu-mate) had a fingerprint reader with a similarly configured common-auth file, and the lock screen on that computer did take a fingerprint before asking for a password.

Boltgolt, 2018-04-26:
You seem to have a talent for discovering rare bugs ;)

I'm quite lost on this one, it might be handling sessions incorrectly? Does `sudo -i` work?

Boltgolt, 2018-04-26:
Also, could you take a look at the last lines in `/var/log/auth.log`?

arifeinberg, 2018-04-26:
sudo -i works fine, but here's auth.log when I log in from the screensaver:
Apr 26 17:58:11 bartlet /lib/security/howdy/pam.py[16770]: Can not open module: Permission denied

Fascinating - I wonder if my user is trying to call the module?

arifeinberg, 2018-04-26:
Changed permissions on pam.py to 750, but then when I went to unlock from screensaver saw this in auth.log:Apr 26 18:01:16 bartlet /lib/security/howdy/pam.py[17152]:   File "/lib/security/howdy/pam.py", line 53, in pam_sm_authenticate
Apr 26 18:01:16 bartlet /lib/security/howdy/pam.py[17152]:     return doAuth(pamh)
Apr 26 18:01:16 bartlet /lib/security/howdy/pam.py[17152]:   File "/lib/security/howdy/pam.py", line 19, in doAuth
Apr 26 18:01:16 bartlet /lib/security/howdy/pam.py[17152]:     if config.get("core", "disabled") == "true":
Apr 26 18:01:16 bartlet /lib/security/howdy/pam.py[17152]:   File "/usr/lib/python2.7/ConfigParser.py", line 607, in get
Apr 26 18:01:16 bartlet /lib/security/howdy/pam.py[17152]:     raise NoSectionError(section)
Apr 26 18:01:16 bartlet /lib/security/howdy/pam.py[17152]: NoSectionError: No section: 'core'

Boltgolt, 2018-04-26:
Both should be impossible, what does your config.ini look like?

arifeinberg, 2018-04-26:
# Howdy config file

[core]
# Do not print anything when a face verification succeeds
no_confirmation = false

# When a user without a known face model tries to use this script, don't
# show an error but fail silently
suppress_unknown = false

# Auto dismiss lock screen on confirmation
# Will run loginctl unlock-sessions after every auth
# Expirimental, can behave incorrectly on some systems
dismiss_lockscreen = false

# Disable howdy in the PAM
# The howdy command will still function
disabled = false 

[video]
# The certainty of the detected face belonging to the user of the account
# On a scale from 1 to 10, values above 5 are not recommended
certainty = 3.5

# The number of seconds to search before timing out
timout = 4

# The /dev/videoX id to capture frames from
# Should be set automatically by the installer
device_id = 0

# Scale down the video feed to this maximum height
# Speeds up face recognition but can make it less precise
max_height = 320

# Because of flashing IR emitters, some frames can be completely unlit
# Skip the frame if the lowest 1/8 of the histogram is above this percentage
# of the total
# The lower this setting is, the more dark frames are ignored
dark_threshold = 70

[debug]
# Show a short but detailed diagnostic report in console
end_report = false

Boltgolt, 2018-05-04:
For anyone having the same issue: @arifeinberg has opened a ticked in the PAM library [here](https://sourceforge.net/p/pam-python/tickets/3/).

sd65, 2018-06-11:
So I have a workaround. Please note I'm using Cinnamon Screensaver (should work on Mate too).

First as you said:
`sudo chmod -R 755 /lib/security/howdy/`

Then:
`sudo howdy config`
And set no_confirmation to `true`.

It's crashing then sending the confirmation message, so disabling it makes the trick. :)

boltgolt, 2018-06-11:
Thanks so much for figuring this out! I'll add it to the FAQ if i ever get around to writing it.

closed 2018-06-11 by boltgolt

papermache, 2019-09-09:
> So I have a workaround. Please note I'm using Cinnamon Screensaver (should work on Mate too).
> 
> First as you said:
> `sudo chmod -R 755 /lib/security/howdy/`
> 
> Then:
> `sudo howdy config`
> And set no_confirmation to `true`.
> 
> It's crashing when sending the confirmation message, so disabling it makes the trick. :)

Can confirm this works on Cinnamon.

djshaji, 2020-01-31:
This does not work on Fedora release 32 (Rawhide) Mate 1.22.3

sudo works, LightDM works, but not the mate lockscreen. I tried making /lib64/security/howdy 0775, but it doesn't work. I've got no /var/log/auth either. Any other ideas?

exander77, 2020-02-18:
Is this secure, to have these files world readable?

boltgolt, 2020-02-25:
Readable shouldn't be a big issue, but don't make it writable

kamushadenes, 2021-01-25:
Can confirm making /lib/security/howdy/ world-readable solved an issue I was facing with `swaylock`.

bananasmoothii, 2021-03-07:
Hi, I got this issue with clearing all faces (`sudo howdy clear`) and creating models with the `-U` flag (`sudo howdy -U <username> add`)

gutierri, 2021-08-31:
This problem appeared to me (Debian 10 + Mate). I believe a fix for cinnamon/mate would be good, the question is whether it is only in these environments (screensavers) that it happens.

I documented this issue on the wiki¹ (common issues).

[1] - https://github.com/boltgolt/howdy/wiki/Common-issues#howdy-pam-module-does-not-load-on-screensaver