boltgolt/howdy #30: ubuntu updated from 17.10 to 18.04

Jerezano, 2018-04-30:
Hello
I updated my laptop from ubuntu 17.10 to 18.04 and I am getting this errors, could you please take a look.

File "/lib/security/howdy/compare.py", line 124, in <module>
    face_encodings = face_recognition.face_encodings(frame)
  File "/usr/local/lib/python3.6/dist-packages/face_recognition/api.py", line 199, in face_encodings
    raw_landmarks = _raw_face_landmarks(face_image, known_face_locations, model="small")
  File "/usr/local/lib/python3.6/dist-packages/face_recognition/api.py", line 153, in _raw_face_landmarks
    face_locations = _raw_face_locations(face_image)
  File "/usr/local/lib/python3.6/dist-packages/face_recognition/api.py", line 102, in _raw_face_locations
    return face_detector(img, number_of_times_to_upsample)
RuntimeError: The stride of the 3rd dimension (the channel dimension) of the numpy array must be 1
Unknown error: 1

Best Regards!

Boltgolt, 2018-04-30:
It looks like the underlying library is having some trouble, might be an outdated version. Could you post what `pip3 show face_recognition` outputs?

Jerezano, 2018-04-30:
Well It should be the latest build from git, I think!

hjerezano:~$ pip3 show face_recognition
Name: face-recognition
Version: 1.2.2
Summary: Recognize faces from Python or from the command line
Home-page: https://github.com/ageitgey/face_recognition
Author: Adam Geitgey
Author-email: ageitgey@gmail.com
License: MIT license
Location: /usr/local/lib/python3.6/dist-packages
Requires: Click, dlib, face-recognition-models, numpy, Pillow
Required-by:

hjerezano:~$ pip3 show numpy
Name: numpy
Version: 1.14.1
Summary: NumPy: array processing for numbers, strings, records, and objects.
Home-page: http://www.numpy.org
Author: NumPy Developers
Author-email: numpy-discussion@python.org
License: BSD
Location: /usr/local/lib/python3.6/dist-packages
Requires:
Required-by: scipy, opencv-python, face-recognition

hjerezano:~$ pip3 show pillow
Name: Pillow
Version: 5.1.0
Summary: Python Imaging Library (Fork)
Home-page: https://python-pillow.org
Author: Alex Clark (Fork Author)
Author-email: aclark@aclark.net
License: Standard PIL License
Location: /usr/lib/python3/dist-packages
Requires:
Required-by: face-recognition

Boltgolt, 2018-05-01:
Strange, i can't find anything about that error anywhere either. It does look like you're using an older version of howdy but that shoudn't really matter.

Could you try commenting out the 
```	
frame = frame[:, :, ::-1]
```
line in `/lib/security/howdy/compare.py`?

Jerezano, 2018-05-01:
Ok did that, and the error is not showing any more!

# Convert from BGR to RGB
#       frame = frame[:, :, ::-1]

In regards the version of howdy before the update of ubuntu I initially had
howdy with github then move to the ppa then updated to ubuntu 18.04
uninstall howdy and reinstalled with synaptic app manager.

Do you think there's any unwanted old files from the initial git?

vigneshgr8, 2018-05-03:
I just replaced the compare.py in my **/lib/security/howdy/** with the compare.py in **src** in the dev branch. _I am new to this so i don't know if thats what should be done._

It retured the following error.

Traceback (most recent call last):
  File "/usr/lib/python3.6/configparser.py", line 789, in get
    value = d[option]
  File "/usr/lib/python3.6/collections/__init__.py", line 883, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/lib/python3.6/collections/__init__.py", line 875, in __missing__
    raise KeyError(key)
KeyError: 'force_mjpeg'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 64, in <module>
    if config.get("debug", "force_mjpeg") == "true":
  File "/usr/lib/python3.6/configparser.py", line 792, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'force_mjpeg' in section: 'debug'
Unknown error: 1

Boltgolt, 2018-05-03:
It would've worked if i didn't add other changes as well. You can just revert to the default `compare.py` and remove the line that reads `frame = frame[:, :, ::-1]`.