boltgolt/howdy #33: Installation failure

ramkrishna757575, 2018-05-10:
```
>>> Installing python dependencies

Collecting face_recognition_models==0.3.0
Requirement already satisfied: Click>=6.0 in /usr/local/lib/python3.6/dist-packages (6.7)
Requirement already satisfied: numpy in /usr/local/lib/python3.6/dist-packages (1.14.3)
Requirement already satisfied: Pillow in /usr/lib/python3/dist-packages (5.1.0)
launchpadlib 1.10.6 requires testresources, which is not installed.
Installing collected packages: face-recognition-models
Successfully installed face-recognition-models-0.3.0

>>> Installing face_recognition

Collecting face_recognition==1.2.2
  Using cached https://files.pythonhosted.org/packages/28/10/f153bbbc218fc169768aa1c02f2e9178e9241e4af8da56289bdca2c0c217/face_recognition-1.2.2-py2.py3-none-any.whl
Installing collected packages: face-recognition
Successfully installed face-recognition-1.2.2

>>> Configuring howdy

Permissions set
Howdy command installed

>>> Adding howdy as PAM module

>>> START OF /etc/pam.d/common-auth
#
# /etc/pam.d/common-auth - authentication settings common to all services
#
# This file is included from other service-specific PAM config files,
# and should contain a list of the authentication modules that define
# the central authentication scheme for use on the system
# (e.g., /etc/shadow, LDAP, Kerberos, etc.).  The default is to use the
# traditional Unix authentication mechanisms.
#
# As of pam 1.0.1-6, this file is managed by pam-auth-update by default.
# To take advantage of this, it is recommended that you configure any
# local modules either before or after the default block, and use
# pam-auth-update to manage selection of other modules.  See
# pam-auth-update(8) for details.

# Howdy IR face recognition
auth	sufficient			pam_python.so /lib/security/howdy/pam.py

# here are the per-package modules (the "Primary" block)
auth	[success=1 default=ignore]	pam_unix.so nullok_secure
# here's the fallback if no module succeeds
auth	requisite			pam_deny.so
# prime the stack with a positive return value if there isn't one already;
# this avoids us returning an error just because nothing sets a success code
# since the modules above will each just jump around
auth	required			pam_permit.so
# and here are more per-package modules (the "Additional" block)
# end of pam-auth-update config
>>> END OF /etc/pam.d/common-auth

Lines will be insterted in /etc/pam.d/common-auth as shown above
Apply this change? [y/N]: y
Inerpeting as a "NO", aborting
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.3-2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
W: Operation was interrupted before it could finish
```

ramkrishna757575, 2018-05-10:
I'm trying to comment the input `Apply this change? [y/N]:`  and want to try to install it from the source.
Can you please help me to run the installation from the source?

Boltgolt, 2018-05-10:
Try removing the half installed package first

```
sudo apt purge howdy
```

then set a flag that will execute the installer without asking any questions

```
HOWDY_NO_PROMPT="true"
```

Then run the installer again. This means howdy will also use the default camera but you can change that with the `sudo howdy config` command afterwards

ramkrishna757575, 2018-05-10:
Okay thanks..I'll try and let u know.

ramkrishna757575, 2018-05-10:
The installation succeeded. But now I'm getting this error on running `sudo howdy add`:

```
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 86, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 9, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
```

Boltgolt, 2018-05-10:
That's an issue on some machines, could you try the solutions from #32?

ramkrishna757575, 2018-05-10:
Thanks. I installed the opencv library and it worked.

ramkrishna757575, 2018-05-10:
It's working great now. You've done an amazing job. Thanks.

Just wanted to know. It uses my webcam for face detection. Is there a way I can configure it to use the IR camera instead?

My system is Alienware 15R3 and has the IR camera.

Boltgolt, 2018-05-10:
Run `sudo howdy config`, which will open the config file for you. Try changing the `device_id` option to 1 and save the file. Howdy should now try to use a different video input.

closed 2018-06-28 by boltgolt