boltgolt/howdy #41: Change to use xed instead of gedit?

stellarpower, 2018-05-18:
I can't use the CLI to open the config as on newer systems gedit's been replaced by xed. IDK if you wanna change this in the source.

Boltgolt, 2018-05-19:
Not sure, other systems might not support both xed and gedit and i don't think nano or other command line editors are user friendly enough. Might have to look into building an actual UI for the config.

stellarpower, 2018-05-19:
For now probably testing for the presence of xed and if not calling gedit
or kate would be a good solution, do you think?

dmig, 2018-06-20:
Solution is simple: use `xdg-open` instead of `gedit` here https://github.com/boltgolt/howdy/blob/84c0501642f761a822d1722f7d4cc395e91c2b6c/src/cli/config.py#L12

dmig, 2018-06-20:
Or, the better solution: use CLI editor `/etc/alternatives/editor`

boltgolt, 2018-06-20:
`xdg-open` opens in a wine version of Notepad for me so i'm not sure about that. Both firefox and nautilus open a .ini in gedit by default for me so i'm not sure what `xdg-open` is using here.

`/etc/alternatives/editor` is nice but nano might be a little confusing for less technical people? I'm sure the amount of people who use Ubuntu and not know how to save in nano is very small, maybe a comment in the config file explaining how to close nano would be enough?

closed 2018-06-28 by boltgolt