boltgolt/howdy #45: Running howdy on Arch Linux

KelleyMcChes, 2018-06-19:
Hey boltgolt,

Earlier, I was curious about getting howdy to run on Fedora and tried to get it running. After it initially didn't work, I just gave up on it. Now that I am on Arch Linux, I retried installing it and finally got it to work with howdy 1.0.0. I don't know how to convert the current deb packages to files that pacman uses, but if you do have an idea, I can give you the current (working) installer.py file. There is no common-auth file in Arch Linux, so you would need to enter the pam.d information into each pam config file.

boltgolt, 2018-06-19:
Thanks for taking the time to look at this! The latest version should probably still be usable if you copy the contents of the [src](https://github.com/boltgolt/howdy/tree/master/src) folder. I'm looking into packaging for arch and fedora when i have the time so please do share your installation script. 

Could i @ mention you if i have a pacman version to test?

KelleyMcChes, 2018-06-20:
After copying all of the files from the current src folder into /lib/security/howdy and manually symlinking the cli.py file into /usr/bin/howdy, the current howdy works. I had to change the /dev/video* to the correct one with `sudo howdy config` but it is perfect.

KelleyMcChes, 2018-06-20:
If you would like me to, I can try packaging it for you

boltgolt, 2018-06-20:
That would be great. Does arch use a separate folder (like `/debian` on apt systems) to store the packaging files? I'm reading the packaging guide and i don't fancy putting the fedora and arch files directly in the root. 

The `/debian/postinst` file contains the current installation script, might be worth looking at.

If you do get a working build, a pull request with the source would be amazing

KelleyMcChes, 2018-06-20:
Well since howdy uses packages from the AUR, your only choice is using the AUR. This will be much easier since you can basically use any format for your source. How the AUR works is there is a file called the PKGBUILD that automates the build and install process. The package is installed in a fakeroot environment and then put into a tar.xz file (Arch's packaging style) and automatically installed. Ignoring a few minor modifications you might need to make when designing the PKGBUILD, it is basically developing a set of instructions to install from source.

boltgolt, 2018-06-20:
Sounds easier than debian, must the PKGBUILD be placed in the root or can the package be build from a subfolder, like `/arch`?

KelleyMcChes, 2018-06-20:
The file is actually stored in git server on the AUR while your files just need to stay in GitHub. [Here](https://aur.archlinux.org/packages/firefox-beta/) is an example of an AUR package.

KelleyMcChes, 2018-06-21:
I'm trying to make the PKGBUILD, but have come across a roadblock. Do you think you could look through the setup.py file in the dlib_clone folder that you create and see if there is anything I can change with `sed` to allow using a DESTDIR while installing. The Arch Linux packaging needs it.

boltgolt, 2018-06-21:
I don't think packaging all of Dlib with Howdy is a very good idea, as the install size is massive. The only reason the installation script does this manual build of Dlib is that the package in the debian repositories tends to fail on installation.

Maybe you can try if arch has a working Dlib build in pacman?

I was also thinking of how we would do installation steps such as configuring the right camera. Debian has these automatic installation scripts but i don't think arch has the same, am i correct? We might need to move that part of the installation to a command like `sudo howdy configure`.

KelleyMcChes, 2018-06-21:
If you look at most packages, Arch has more if a "install and then configure" style. For example, when installing the grub package, you are expected to do the installation to the efi partition afterwards ([see here](https://wiki.archlinux.org/index.php/GRUB)). I think the best route would be to leave the postinstallation configuration to the user and make a simple wiki page on the Archwiki explaining how to install and configure the package.

With dlib, I can check whether the installation of dlib on Arch included the things that howdy needs, but I was only going to have the python part of dlib be included with the package. The issue is that a proper Arch package has the final installation of a package (when you would run `sudo make install`) have a prefix of /usr (which doesn't apply for this dlib python API) and a destdir of $pkgdir/ ([see here](https://wiki.archlinux.org/index.php/Creating_packages#package.28.29))

KelleyMcChes, 2018-06-23:
Just so you know, I have finished creating the PKGBUILD and tested it, so I know it's working. I checked to see if the dlib Python API was in the Arch repositories and it didn't seem to be there, so I built it (the proper way according to the ArchWiki). If you want to look (and if you could, possibly check if it works in a live USB of Arch?), I have attached the file. If you're alright with it, could you add yourself as a maintainer and send me the file so I can upload it? If you're alarmed about my name being in the maintainer comment also, the maintainer is for both the package and the PKGBUILD.
[PKGBUILD.tar.gz](https://github.com/boltgolt/howdy/files/2129665/PKGBUILD.tar.gz)

ataicl, 2018-06-23:
hello, I test the PKGBUILD,  and I get this errors:

<code>
$makepkg
==> Making package: howdy 2.2.2-1 (Sat 23 Jun 2018 03:46:47 PM CST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found v2.2.2.tar.gz
  -> Found pam-python-1.0.6.tar.gz
  -> Found pam-python-1.0.6-fedora.patch
  -> Found pam-python-1.0.6-gcc8.patch
==> Validating source files with sha256sums...
    v2.2.2.tar.gz ... Passed
    pam-python-1.0.6.tar.gz ... Passed
    pam-python-1.0.6-fedora.patch ... Passed
    pam-python-1.0.6-gcc8.patch ... Passed
==> Extracting sources...
  -> Extracting v2.2.2.tar.gz with bsdtar
  -> Extracting pam-python-1.0.6.tar.gz with bsdtar
==> Starting prepare()...
Cloning into 'dlib_clone'...
remote: Counting objects: 2277, done.
remote: Compressing objects: 100% (2222/2222), done.
remote: Total 2277 (delta 372), reused 777 (delta 43), pack-reused 0
Receiving objects: 100% (2277/2277), 10.56 MiB | 439.00 KiB/s, done.
Resolving deltas: 100% (372/372), done.
patching file src/pam_python.c
Hunk #1 succeeded at 39 (offset 7 lines).
Hunk #2 succeeded at 1526 (offset 29 lines).
patching file src/pam_python.c
Hunk #1 succeeded at 2227 (offset 53 lines).
patching file src/Makefile
Hunk #1 succeeded at 1 with fuzz 1.
==> Starting build()...
make --directory doc
make --directory src
make[1]: Entering directory '/tmp/makepkg/howdy/src/pam-python-1.0.6/doc'
sphinx-build2 -b html -E . html
make[1]: Entering directory '/tmp/makepkg/howdy/src/pam-python-1.0.6/src'
gcc -O0 -Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wsign-compare -Waggregate-return -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Werror -Wno-error=cast-function-type -g -o ctest ctest.c -lpam
sed "s,\\\$PWD,$(pwd),g" "test-pam_python.pam.in" >"test-pam_python.pam.tmp" 
In file included from /usr/include/link.h:23,
                 from ctest.c:14:
/usr/include/features.h:428:12: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
            ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:29: ctest] Error 1
make[1]: *** Waiting for unfinished jobs....
mv test-pam_python.pam.tmp test-pam_python.pam
CFLAGS="-Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wsign-compare -Waggregate-return -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Werror -Wno-error=cast-function-type -I/usr/local/lib/ " ./setup.py build
/usr/lib/python2.7/pkgutil.py:183: ImportWarning: Not importing directory '/usr/lib/python2.7/site-packages/sphinxcontrib': missing __init__.py
  file, filename, etc = imp.find_module(subname, path)
/usr/lib/python2.7/pkgutil.py:183: ImportWarning: Not importing directory '/usr/lib/python2.7/site-packages/jaraco': missing __init__.py
  file, filename, etc = imp.find_module(subname, path)
running build
running build_ext
building 'pam_python' extension
creating build
creating build/temp.linux-x86_64-2.7
gcc -pthread -DNDEBUG -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wsign-compare -Waggregate-return -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Werror -Wno-error=cast-function-type -I/usr/local/lib/ -D_FORTIFY_SOURCE=2 -fPIC -DLIBPYTHON_SO="libpython2.7.so.1.0" -I/usr/include/python2.7 -c pam_python.c -o build/temp.linux-x86_64-2.7/pam_python.o
In file included from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdlib.h:25,
                 from /usr/include/security/_pam_macros.h:12,
                 from pam_python.c:33:
/usr/include/features.h:428:12: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
            ^~~~~~~~~~~~~
compilation terminated.
error: command 'gcc' failed with exit status 1
make[1]: *** [Makefile:11: pam_python.so] Error 1
make[1]: Leaving directory '/tmp/makepkg/howdy/src/pam-python-1.0.6/src'
make: *** [Makefile:6: lib] Error 2
make: *** Waiting for unfinished jobs....
Running Sphinx v1.7.5
making output directory...
loading intersphinx inventory from http://docs.python.org/2.7/objects.inv...
intersphinx inventory has moved: http://docs.python.org/2.7/objects.inv -> https://docs.python.org/2.7/objects.inv
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 1 source files that are out of date
updating environment: 1 added, 0 changed, 0 removed
reading sources... [100%] pam_python                                            
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
writing output... [100%] pam_python                                             
generating indices... genindex
writing additional pages... search
copying static files... done
copying extra files... done
dumping search index in English (code: en) ... done
dumping object inventory... done
build succeeded.

The HTML pages are in html.
rm -f html/index.html && ln -s pam_python.html html/index.html
make[1]: Leaving directory '/tmp/makepkg/howdy/src/pam-python-1.0.6/doc'
==> ERROR: A failure occurred in build().
    Aborting...
</code>

boltgolt, 2018-06-23:
PKGBUILD looks good as far as i can see, but i don't have the time right now to get a arch installation going. If it works for you locally and @ataicl's issue is resolved i'd say you could release it.

As for the maintainer field: It's only fitting to put you in there as a maintainer as well, you've done some stellar work so far.  You can use `boltgolt <boltgolt@gmail.com>` for my field so it lines up with the debian release.

ataicl, 2018-06-24:
thank you! Because my notebook is ryzen apu, so I use linux-mainline

KelleyMcChes, 2018-06-24:
@ataicl 
Could you try installing lib32-glibc and retry? It might need a 32-bit library.

ataicl, 2018-06-24:
I installed lib32-glibc , and I got a similar error：

==========
$makepkg
==> Making package: howdy 2.2.2-1 (Sun 24 Jun 2018 03:25:40 PM CST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found v2.2.2.tar.gz
  -> Found pam-python-1.0.6.tar.gz
  -> Found pam-python-1.0.6-fedora.patch
  -> Found pam-python-1.0.6-gcc8.patch
==> Validating source files with sha256sums...
    v2.2.2.tar.gz ... Passed
    pam-python-1.0.6.tar.gz ... Passed
    pam-python-1.0.6-fedora.patch ... Passed
    pam-python-1.0.6-gcc8.patch ... Passed
==> Extracting sources...
  -> Extracting v2.2.2.tar.gz with bsdtar
  -> Extracting pam-python-1.0.6.tar.gz with bsdtar
==> Starting prepare()...
Cloning into 'dlib_clone'...
remote: Counting objects: 2277, done.
remote: Compressing objects: 100% (2222/2222), done.
remote: Total 2277 (delta 372), reused 799 (delta 43), pack-reused 0
Receiving objects: 100% (2277/2277), 10.56 MiB | 2.52 MiB/s, done.
Resolving deltas: 100% (372/372), done.
patching file src/pam_python.c
Hunk #1 succeeded at 39 (offset 7 lines).
Hunk #2 succeeded at 1526 (offset 29 lines).
patching file src/pam_python.c
Hunk #1 succeeded at 2227 (offset 53 lines).
patching file src/Makefile
Hunk #1 succeeded at 1 with fuzz 1.
==> Starting build()...
make --directory doc
make --directory src
make[1]: Entering directory '/tmp/makepkg/howdy/src/pam-python-1.0.6/doc'
sphinx-build2 -b html -E . html
make[1]: Entering directory '/tmp/makepkg/howdy/src/pam-python-1.0.6/src'
gcc -O0 -Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wsign-compare -Waggregate-return -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Werror -Wno-error=cast-function-type -g -o ctest ctest.c -lpam
sed "s,\\\$PWD,$(pwd),g" "test-pam_python.pam.in" >"test-pam_python.pam.tmp" 
In file included from /usr/include/link.h:23,
                 from ctest.c:14:
/usr/include/features.h:428:12: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
            ^~~~~~~~~~~~~
compilation terminated.
mv test-pam_python.pam.tmp test-pam_python.pam
make[1]: *** [Makefile:29: ctest] Error 1
make[1]: *** Waiting for unfinished jobs....
CFLAGS="-Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wsign-compare -Waggregate-return -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Werror -Wno-error=cast-function-type -I/usr/local/lib/ " ./setup.py build
/usr/lib/python2.7/pkgutil.py:183: ImportWarning: Not importing directory '/usr/lib/python2.7/site-packages/sphinxcontrib': missing __init__.py
  file, filename, etc = imp.find_module(subname, path)
/usr/lib/python2.7/pkgutil.py:183: ImportWarning: Not importing directory '/usr/lib/python2.7/site-packages/jaraco': missing __init__.py
  file, filename, etc = imp.find_module(subname, path)
running build
running build_ext
building 'pam_python' extension
creating build
creating build/temp.linux-x86_64-2.7
gcc -pthread -DNDEBUG -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -Wall -Wextra -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wsign-compare -Waggregate-return -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Werror -Wno-error=cast-function-type -I/usr/local/lib/ -D_FORTIFY_SOURCE=2 -fPIC -DLIBPYTHON_SO="libpython2.7.so.1.0" -I/usr/include/python2.7 -c pam_python.c -o build/temp.linux-x86_64-2.7/pam_python.o
In file included from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdlib.h:25,
                 from /usr/include/security/_pam_macros.h:12,
                 from pam_python.c:33:
/usr/include/features.h:428:12: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
            ^~~~~~~~~~~~~
compilation terminated.
error: command 'gcc' failed with exit status 1
make[1]: *** [Makefile:11: pam_python.so] Error 1
make[1]: Leaving directory '/tmp/makepkg/howdy/src/pam-python-1.0.6/src'
make: *** [Makefile:6: lib] Error 2
make: *** Waiting for unfinished jobs....
Running Sphinx v1.7.5
making output directory...
loading intersphinx inventory from http://docs.python.org/2.7/objects.inv...
intersphinx inventory has moved: http://docs.python.org/2.7/objects.inv -> https://docs.python.org/2.7/objects.inv
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 1 source files that are out of date
updating environment: 1 added, 0 changed, 0 removed
reading sources... [100%] pam_python                                            
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
writing output... [100%] pam_python                                             
generating indices... genindex
writing additional pages... search
copying static files... done
copying extra files... done
dumping search index in English (code: en) ... done
dumping object inventory... done
build succeeded.

The HTML pages are in html.
rm -f html/index.html && ln -s pam_python.html html/index.html
make[1]: Leaving directory '/tmp/makepkg/howdy/src/pam-python-1.0.6/doc'
==> ERROR: A failure occurred in build().
    Aborting...

KelleyMcChes, 2018-06-24:
@ataicl How about this?
`pkgfile /usr/include/sys/cdefs.h core/glibc`

ataicl, 2018-06-24:
@KelleyMcChes   I installed glibc and cmake, and i Successful compilation。 thank you !

But, when i pacman -U , I got error:

======
#pacman -U howdy-2.2.2-1-x86_64.pkg.tar.xz 
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) howdy-2.2.2-1

Total Installed Size:  202.61 MiB

:: Proceed with installation? [Y/n] 
(1/1) checking keys in keyring                     [######################] 100%
(1/1) checking package integrity                   [######################] 100%
(1/1) loading package files                        [######################] 100%
(1/1) checking for file conflicts                  [######################] 100%
error: failed to commit transaction (conflicting files)
howdy: /lib exists in filesystem (owned by filesystem)
Errors occurred, no packages were upgraded.

KelleyMcChes, 2018-06-24:
@ataicl I have attached a new PKGBUILD. Please try running this one in a clean directory.
[PKGBUILD.tar.gz](https://github.com/boltgolt/howdy/files/2130932/PKGBUILD.tar.gz)

KelleyMcChes, 2018-06-25:
@boltgolt I have created a wiki page [here](https://wiki.archlinux.org/index.php/Howdy) for Howdy and have [uploaded it to the AUR](https://aur.archlinux.org/packages/howdy/). If you would like, you can create an account on the AUR and I can add you as a co-maintainer.

boltgolt, 2018-06-25:
Nice work! I've created an AUR account with the same password as the maintainer field in the PKGBUILD.

If you want to update the readme with a link to the wiki (and maybe dropping the "Ubuntu" in the title) feel free to open a pull request. Otherwise i will update it with the 2.3.0 release in a couple of weeks.

closed 2018-06-25 by KelleyMcChes