boltgolt/howdy #50: howdy fails on arch

ChrisChinh, 2018-07-03:
Hello,
please excuse any incorrect formatting, this is my first time using Github. I installed howdy using the aur package but when I run `sudo howdy test` I get:
`Unable to stop the stream: Invalid argument

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Traceback (most recent call last):
  File "/usr/bin/howdy", line 98, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 78, in <module>
    overlay = frame.copy()
AttributeError: 'NoneType' object has no attribute 'copy'
`
Am I doing something wrong?
Thanks in advance for the support

boltgolt, 2018-07-03:
@KelleyMcChes problemens with the cv2 python bindings perhaps?

KelleyMcChes, 2018-07-03:
There are currently a few problems with the Arch package. I am working on it currently and should have it fixed in the next few days. @boltgolt @ChrisChinh

boltgolt, 2018-07-03:
Great, thanks!

KelleyMcChes, 2018-07-20:
I finally got howdy updated and the issue should be fixed. Don't forget to change /dev/videoX in the config file or you will get an error!

boltgolt, 2018-07-20:
Great, thanks! 

Is it possible for a multi-distro package to track the `.SRCINFO` and `PKGBUILD` files within the main repo? I find it a bit unnerving that such important files are not saved in the repository and are only available on AUR. I guess there is a chicken-egg problem with the sha256sums but that's only a small part of the files.

KelleyMcChes, 2018-07-20:
The problem is that the .SRCINFO is generated dynamically with every new PKGBUILD. If there ever seems to be an issue with a new version and I have to modify the PKGBUILD by adding a package or something similar, the .SRCINFO will change and thus the sha256 of the package will change. There's a reason no package stores it's checksums inside the downloaded zip or tar file :wink:

Madseason41, 2018-08-14:
I have the same issue when I update the kernel on Ubuntu 18.04 to 4.18 from 4.15.
Once i revert back to 4.15 everything works fine.

Error when using 4.18 and trying sudo apt update:

Unable to stop the stream: Invalid argument
VIDEOIO ERROR: V4L: can't open camera by index 1
Face detection timeout reached

Error when trying sudo howdy test:

Unable to stop the stream: Invalid argument
VIDEOIO ERROR: V4L: can't open camera by index 1

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 102, in <module>
    import cli.test
  File "/lib/security/howdy/cli/test.py", line 85, in <module>
    overlay = frame.copy()
AttributeError: 'NoneType' object has no attribute 'copy'

Ideas?

boltgolt, 2018-10-04:
Following up on issues i forgot to reply to in august, are you still having this problem or did you fix it/uninstall Howdy?

closed 2018-10-24 by boltgolt