boltgolt/howdy #51: IR emitter blink only with specific resolution

janecz-n, 2018-07-10:
Hi everyone,

I had write an issues few month ago: https://github.com/boltgolt/howdy/issues/19
I was a bit curious about why emitter emit when I used **fswebcam** but not when I used Cheese or Howdy. I play with it today and notice something:

**By default fswebcam adjust the resolution from 384x288 to 340x340**

I had try to use the parameter -r to set another resolution and it adjust resolution from 1280x720 to 640x480 and the emitter didn't work.

Where you aware of this issue ? 

I also try to set Cheese in 340x340 and got this as a result, has I never used IR camera, is this a valid output ? http://webmshare.com/play/QNzyr

P.S: I'm on Ubuntu 18.04

boltgolt, 2018-07-10:
Since your first issue i've seen a few of these weird resolution problems. For this reason i introduced the `frame_width` and `frame_height` config options in the latest release, you might want to check those out. For some people only the largest resolution works (like for me), for some you need to smaller one. There's not great default here unfortunately. 

Your output is completely valid. The flashing if the LEDs are never in sync and produce this effect where about half of the frames are dark. Howdy automatically ignores these dark frames, as you can see with `sudo howdy test`.

janecz-n, 2018-07-10:
Perfect, thanks for the answer.
I will try to make it work :)

closed 2018-07-10 by janecz-n

Zocker1999NET, 2019-06-06:
I just tried to use howdy on my Thinkpad X1 Tablet Gen3 and had the same issue,
fixed it using the resolution 340x340.

But however, it took too long to discover this issue.
@boltgolt Would you please add a good description of this issue into the README file,
so that people having the same issue / using the same devices as we can find this fix more easily?

boltgolt, 2019-06-09:
I've added a [wiki page](https://github.com/boltgolt/howdy/wiki/Common-issues#camera-ir-emitters-not-turning-on) for common issues and also liked to it in the readme