boltgolt/howdy #53: IR emitter will sometimes not stop firing after authenticating

Ethiarpus, 2018-07-13:
This seems to apply to all applications except the terminal and the login screen.
For example, if I try to use Synaptic or GParted or elevate Nemo to root, the IR emitter will fire and authenticate, but then it will continue to fire for several minutes afterwards, before stopping on its own. If you try to run another authentication in the terminal it will throw `VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV
Unable to stop the stream: Device or resource busy` and in a gui it will say "Face detection timeout reached" and re-show the password box.

----

Running Linux Mint 18.3 Cinnamon 64-Bit
Running Howdy 2.3.0

boltgolt, 2018-07-13:
The really weird part is that it occurs everywhere except on the login screen or in the terminal, which lets me to believe it's an openCV bug.

Could you give me your python3 version? (`python3 -V`) Also, did you install Howdy with the apt command?

closed 2018-08-23 by boltgolt

boltgolt, 2019-01-22:
Do any errors show up in the terminal? Are there any errors in `/var/log/auth.log`?