boltgolt/howdy #54: howdy test segment violation

zago88, 2018-07-14:
On excute howdy test fail with output

>Opening a window with a test feed

>Press ctrl+C in this terminal to quit
>Click on the image to enable or disable slow mode

>QObject::moveToThread: Current thread (0x8fc4420) is not the object's thread (0x91c03a0).
>Cannot move to target thread (0x8fc4420)

>Violación de segmento  

----

**Kubuntu 18.04**:
**Howdy 2.3.0**:
**Python 3.6.5**.

zago88, 2018-07-19:
Thanks for the answer, This exceed my knowledge since the problem is the opencv library directly,  I'll try to investigate how to patch this issue with the namespace on Qt platform

closed 2018-08-23 by boltgolt

antonio-petricca, 2020-06-16:
Hi, I have the same issue on Linux Mint 19.3 (Ubuntu 18.04).

Any solution yet?