boltgolt/howdy #55: Howdy support for IR camera in Lenovo Thinkpad T480s

thuanxt, 2018-07-15:
I've installed howdy in my laptop Lenovo Thinkpad T480s with the camera details as the following link:
https://github.com/boltgolt/howdy-reports/issues/14

After the installation, I ran 'sudo howdy add' but it didn't work. The traceback is as below:
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 90, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 9, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
----
**Linux distribution (if applicable)**: Ubuntu 18.04
**Howdy version**: v2.3.0

Could you please help me to resolve the problem. Let me know if you need further information.
Thank you,
Ryan

boltgolt, 2018-07-15:
Thanks for opening an issue. This is an known problem, take a look at #32.

closed 2018-07-15 by boltgolt

thuanxt, 2018-07-16:
After installing the opencv-python module, I can run the command to add face model but facing the error "No face detected, aborting". I tried to run 'sudo howdy test' command and a window open which shows nothing so I think the camera cannot detect my face, the IR camera was turning on though (there are 2 red led lights which indicating the IR camera is turning on).

Please refer to the following image for the testing window.
[image]

thuanxt, 2018-07-16:
Thank you, I can scan and add a face model right now. However, it seems that the camera cannot recognize my face after adding models (I have added a lot of models though). It always shows "Face detection timeout reached". Is there any configurations to decrease the acceptable accuracy for matching face model. Also be noted that security is not my concern, I just don't want to type password whenever running a command as a root user.

boltgolt, 2018-07-16:
You can try upping the `certainty` config option to something higher. This will also result in more false positives, but should be fine if security isn't your main concern.

thuanxt, 2018-07-16:
Thank you for your quick response and help. I've increased the `certainty` config option to 10 with no luck. Is there any ways which I can view the matching result in order to figure out what wrong with my camera?

thuanxt, 2018-07-17:
After moving my face a little far away from the camera, it is working. Thank you very much for your support.

boltgolt, 2018-07-17:
Good to hear! Be sure to limit the amount of models you save, every additional model you add slows down recognition.