boltgolt/howdy #58: permission denied

reverland, 2018-08-01:
_Please describe the issue in as much detail as possible, including any errors and traces_
when using sudo, give out permission error

```
➜   sudo bash
python3: can't open file '/lib/security/howdy/compare.py': [Errno 13] Permission denied
Unknown error: 2
```

----

**Linux distribution (if applicable)**: ubuntu 18.04
**Howdy version**: 2.3.0 from ppa

reverland, 2018-08-01:
I workaround this by :
1. chmod 644 /lib/security/howdy/{compare.py,config.ini}
2.  change /lib/security/howdy/pam.py       
```
status = subprocess.call(["/usr/bin/python3", os.path.dirname(os.path.abspath(__file__)) + "/compare.py", pamh.get_user()])
```

boltgolt, 2018-08-02:
PAM is running as root by default, but i've seen a few of similar issues where PAM is running Howdy without root rights. Is Howdy fully functional now you've changed the permissions of only those 2 files?

reverland, 2018-08-12:
yeah，looks like its fully functional.

I use it when login and type `sudo` in gnome terminal only, however.

reverland, 2018-08-12:
```
➜  ~ sudo howdy test

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Closing window
```

nothing strange happened

reverland, 2018-08-12:
but I see nothing

reverland, 2018-08-12:
I use pyenv, so the python path is different, maybe this issue should be split into 2 issues focus on different things

boltgolt, 2018-10-04:
Following up on issues i forgot to reply to, are you still having this problem or did you fix it/uninstall Howdy?

reverland, 2018-10-15:
I did my workaround and it works well for a long long long long time.

I think the workaround may be necessary.

boltgolt, 2018-10-24:
A fix is now planned for the next release, main threat is #83

closed 2018-10-24 by boltgolt