boltgolt/howdy #59: Using with Linux Mint

Rhiyo, 2018-08-02:
Hey there, I'm new to linux as you can probably tell by the distro I'm using.

I'm trying to use Howdy with Linux mint. I've got Howdy working with the IR cameras as I'm able to use commands and add new faces and test out the IR-Camera but I haven't been able to get it to actually work during sudo log in, the lock screen, or beginning of computer sign in.

I'm I supposed to integrate it myself using PAM somehow, or am is this related to using Linux Mint maybe?

I apologize if I've missed something obvious.

Thank you!

boltgolt, 2018-08-04:
It should be enabled in PAM automatically, what does your common-auth file look like? (`sudo cat /etc/pam.d/common-auth`) 

When Howdy encounters an error, it tends to fail silently because that's recommended for PAM. modules. You could check `/var/log/auth.log` for any errors?

boltgolt, 2018-08-05:
It seems that Howdy didn't get inserted into common-auth during install, which is really weird. Do you recall being asked if the proposed changes to this file should be applied (during the apt install howdy command)?

Either way, try adding 
```
auth	sufficient			pam_python.so /lib/security/howdy/pam.py
```
on the empty line between "# pam-auth-update(8) for details." and "# here are the per-package modules (the "Primary" block)". You can use nano (`sudo nano /etc/pam.d/common-auth`) for this if you didn't know already.

Rhiyo, 2018-08-05:
Thanks, that worked! I don't remember having that option during install but I may have made a mistake. Now I have a different issue though, somehow it's using my normal camera instead of my IR camera. It originally was using my IR camera (I chose it correctly during set up and was able to see it working when using the test command.)

Rhiyo, 2018-08-05:
Yup, both are using the normal webcam camera. I'm using a Ryzen HP x360 Envy (13.3inch).

Also, it's currently not working on the lock screen. I have to go back to the user selection screen for it to start trying face recognition, Is that the intention? Sorry to bombard you with questions.

boltgolt, 2018-08-06:
Try running `sudo howdy config`  and changing the device id value (probably a number between 0 and 4) until you get the right camera. There are more sophisticated ways to determine which one is the IR camera, but trust me, this is a lot quicker.

As for the lock screen issue: If i recall correctly there is a permission issue sometimes. Could you try the steps in [this comment](https://github.com/boltgolt/howdy/issues/28#issuecomment-396231042)?

Rhiyo, 2018-08-06:
Everything appears to working fine now! Thankyou so much for all your help and this lovely tool!

boltgolt, 2018-08-06:
You're welcome!

closed 2018-08-06 by boltgolt