boltgolt/howdy #65: Support for gnome-screensaver

MartLugt, 2018-08-26:
I currently can't seem to get howdy to work with gnome-screensaver. 
I think this would be a great feature and it would make howdy even more like Windows Hello.

boltgolt, 2018-08-26:
It should work with anything requiring a password, what distro are you using?

MartLugt, 2018-08-26:
I am using Arch

boltgolt, 2018-08-26:
Is anything logged in `/var/log/auth.log` when locking and unlocking?

MartLugt, 2018-08-26:
`/var/log/auth.log` doesn't seem to exist on my system, but using journalctl I found the following:

```
Aug 26 21:37:22 MLArch /lib/security/howdy/pam.py[3530]: Can not open module: Permission denied  
Aug 26 21:37:22 MLArch gnome-screensaver-dialog[3530]: pam_tally(gnome-screensaver:auth): Error opening /var/log/faillog for update  
Aug 26 21:37:22 MLArch gnome-screensaver-dialog[3530]: pam_tally(gnome-screensaver:auth): Error opening /var/log/faillog for read  
Aug 26 21:37:24 MLArch gnome-screensaver-dialog[3530]: gkr-pam: the password for the login keyring was invalid.  
Aug 26 21:37:24 MLArch /lib/security/howdy/pam.py[3530]: Can not open module: Permission denied  
Aug 26 21:37:24 MLArch gnome-screensaver-dialog[3530]: pam_tally(gnome-screensaver:setcred): Error opening /var/log/faillog for update  
Aug 26 21:37:24 MLArch gnome-screensaver-dialog[3530]: pam_tally(gnome-screensaver:setcred): Error opening /var/log/faillog for update  
Aug 26 21:37:24 MLArch gnome-screensaver-dialog[3530]: pam_env(gnome-screensaver:setcred): Unrecognized Option: -Dswt.autoScale=125
                                                        - ignoring line
```

After checking the permissions on `/lib/security/howdy` I found that the directory and all of the files in it were only visible to the owner, root. (According to the README it is best to keep howdy read-only. I think I made a mistake with setting the permissions here).

After resetting the permissions howdy does indeed start up, but it doesn't log me in. It just prints `Identified face as ...`, but still requires me to type in my password.

boltgolt, 2018-08-26:
That's troubling, i'm guessing that nothing gets logged when it only prints that? Is login in after a reboot still working?

MartLugt, 2018-08-26:
After a reboot I can login through Howdy in the virtual terminal.

I did find that an error does get logged:

```
Traceback (most recent call last):
    File "/lib/security/howdy/pam.py", line 53, in pam_sm_authenticate
        return doAuth(pamh)
    File "/lib/security/howdy/pam.py", line 38, in doAuth
        pamh.conversation(pamh.Message(pamh.PAM_TEXT_INFO, "Identified face as " + pamh.get_user()))
PamException: Conversation error
```

I suspect that even though Howdy prints that the face is identified, but then throws this error, causing Pam to see the authentication as failed.

MartLugt, 2018-08-26:
That did the trick! Thanks alot!

closed 2018-08-26 by MartLugt

DragonAxe, 2019-01-22:
@MartLugt What did you reset the permissions to?
Here is what they are currently set to for me. I haven't changed any of these permissions.

    $ sudo ls -la /lib/security/howdy
    total 36
    drw------- 4 root root 4096 Jan 21 14:44 .
    drwxr-xr-x 4 root root 4096 Jan 21 14:30 ..
    drw------- 3 root root 4096 Jan 21 14:44 cli
    -rwx--x--x 1 root root 3314 Jan 21 14:27 cli.py
    -rw------- 1 root root 5413 Jan 21 14:27 compare.py
    -rw------- 1 root root 1764 Jan 21 15:28 config.ini
    drwxr-xr-x 2 root root 4096 Jan 21 14:47 models
    -rw------- 1 root root 2599 Jan 21 14:27 pam.py

I have a Thinkpad T480 running Arch linux with the `i3` window manager. Howdy works with `sudo` and `lightdm`, but not `gnome-screensaver`. With `gnome-screensaver`, the IR lights don't even try to light up.

I run `gnome-screensaver` as my user (not as root). I don't know if that makes a difference.
I lock my screen using `gnome-screensaver-command -l`.

Journalctl shows this after locking and unlocking my screen:

    $ journalctl --since="2019-01-21 18:28:00"
    Jan 21 18:28:57 Satellite-L /lib/security/howdy/pam.py[18352]: Can not open module: Permission denied
    Jan 21 18:28:57 Satellite-L gnome-screensaver-dialog[18352]: pam_tally(gnome-screensaver:auth): Error opening /var/log/faillog for update
    Jan 21 18:28:57 Satellite-L gnome-screensaver-dialog[18352]: pam_tally(gnome-screensaver:auth): Error opening /var/log/faillog for read
    Jan 21 18:29:02 Satellite-L /lib/security/howdy/pam.py[18352]: Can not open module: Permission denied
    Jan 21 18:29:02 Satellite-L gnome-screensaver-dialog[18352]: pam_tally(gnome-screensaver:setcred): Error opening /var/log/faillog for update
    Jan 21 18:29:02 Satellite-L gnome-screensaver-dialog[18352]: pam_tally(gnome-screensaver:setcred): Error opening /var/log/faillog for update

I have set `no_confirmation = true` in `/lib/security/howdy/config.ini`.
I have added `auth sufficient pam_python.so /lib/security/howdy/pam.py` to `/etc/pam.d/gnome-screensaver`, `/etc/pam.d/sudo`, and `/etc/pam.d/lightdm`.

    $ sudo cat /etc/pam.d/gnome-screensaver
    #%PAM-1.0
    auth sufficient pam_python.so /lib/security/howdy/pam.py
    auth        substack     system-login
    -auth        optional     pam_gnome_keyring.so