boltgolt/howdy #68: Face scan in the dark

MorNando, 2018-09-07:
When using my laptop in the dark it doesn't see my face. My camera does have this capability.

It would be fantastic if this could be added at some point.

boltgolt, 2018-09-07:
It does have the capability, but chances are that you're using the visible light camera and not the required infra red one. Could you run `sudo howdy test` and see if the video feed is in black and white or in full color?

MorNando, 2018-09-07:
Thanks for the quick reply. I will check tonight and get back to you.

MorNando, 2018-09-07:
I have just ran `sudo howdy test` and it is indeed in colour. Therefore it is using the visible light camera.

I ran lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 5: Dev 2, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 5: Dev 2, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 3, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 7: Dev 3, If 1, Class=Wireless, Driver=btusb, 12M

then lsusb -v (see attached)
[lsusb.txt](https://github.com/boltgolt/howdy/files/2362545/lsusb.txt)

Thanks in advance.

boltgolt, 2018-09-14:
@JoakimJoensuu that's a completely separate issue, please take a look at #23

boltgolt, 2018-09-14:
@MorNando i've seen this issue before, sometimes the IR camera just won't be correctly picked up in windows and i don't have the knowledge required to figure out why. In your lsusb output it correctly states `HP TrueVision FHD RGB-IR`, but it just doesn't show of as v4l device.

MorNando, 2018-09-14:
Thanks for the reply @boltgolt

If I or someone else figures this out I will post the solution on here for you.

dmig, 2018-12-09:
Probably you have a camera capable of visible and IR modes. Try to play with its v4l controls.
You can use `v4l2-ctrl` or VLC in capture mode (tools -> effects and filters -> v4l2 controls).

boltgolt, 2019-01-06:
I'm closing this for now because it's inactive, but don't be afraid to comment on it if you're also affected by this problem

closed 2019-01-06 by boltgolt