boltgolt/howdy #71: compiling shape_predictor.cpp.o

lcseguin, 2018-09-18:
[ 89%] Building CXX object CMakeFiles/dlib_python.dir/src/shape_predictor.cpp.o
In file included from /tmp/dlib_clone/dlib/../dlib/matrix.h:11:0,
                 from /tmp/dlib_clone/dlib/../dlib/python/numpy_image.h:8,
                 from /tmp/dlib_clone/dlib/../dlib/python.h:9,
                 from /tmp/dlib_clone/tools/python/src/rectangles.cpp:4:
/tmp/dlib_clone/dlib/../dlib/matrix/matrix_la.h: In function ‘long int dlib::svd4(dlib::svd_u_mode, bool, const dlib::matrix_exp<EXP>&, dlib::matrix<typename EXP::type, uM, uN, MM1, L1>&, dlib::matrix<typename EXP::type, qN, qX, MM2, L1>&, dlib::matrix<typename EXP::type, vM, vN, MM3, L1>&) [with EXP = dlib::matrix_op<dlib::op_trans<dlib::matrix_op<dlib::op_trans<dlib::matrix<double, 1, 2> > > > >; long int qN = 1; long int qX = 1; long int uM = 1; long int uN = 1; long int vM = 2; long int vN = 1; MM1 = dlib::memory_manager_stateless_kernel_1<char>; MM2 = dlib::memory_manager_stateless_kernel_1<char>; MM3 = dlib::memory_manager_stateless_kernel_1<char>; L1 = dlib::row_major_layout]’:
/tmp/dlib_clone/dlib/../dlib/matrix/matrix_la.h:225:32: warning: iteration 1 invokes undefined behavior [-Waggressive-loop-optimizations]
             y = abs(q(i)) + abs(e(i));
                             ~~~^~~~~~
/tmp/dlib_clone/dlib/../dlib/matrix/matrix_la.h:163:20: note: within this loop
         for (i=0; i<n; i++)
                   ~^~

lcseguin, 2018-09-20:
I forgot to mention that after this error, the compiling continue and when done, Howdy seem's working fine.
Notice : before installing, i've installed the python3-opencv package.

boltgolt, 2018-09-20:
Very good to hear! I'll close it for now as it's not fatal to the installation, if anyone also has this issue feel free to reopen. Thanks for reporting!

closed 2018-09-20 by boltgolt