boltgolt/howdy #76: VIDEOIO ERROR: V4L: can't open camera by index 3

RogerBergling, 2018-09-26:
The camera is working with other tools like Cheese
Dell XPS 13 9370 
v4l2-ctl --list-devices
Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-5):

T:  Bus=01 Lev=01 Prnt=01 Port=04 Cnt=02 Dev#=  2 Spd=480 MxCh= 0
D:  Ver= 2.01 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=58f4 Rev=72.79
S:  Manufacturer=CN0FFMHCLOG0083EB4B2A01
S:  Product=Integrated_Webcam_HD
S:  SerialNumber=200901010001
C:  #Ifs= 4 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=01 Driver=uvcvideo
I:  If#= 1 Alt= 0 #EPs= 0 Cls=0e(video) Sub=02 Prot=01 Driver=uvcvideo
I:  If#= 2 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=01 Driver=uvcvideo
I:  If#= 3 Alt= 0 #EPs= 0 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo

howdy -U test -y add
Adding face model for the user test
Using default label "Initial model" because of -y flag
Unable to stop the stream: Invalid argument
VIDEOIO ERROR: V4L: can't open camera by index 3

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 90, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 117, in <module>
    enc = face_recognition.face_encodings(frame)
  File "/usr/local/lib/python3.6/dist-packages/face_recognition/api.py", line 199, in face_encodings
    raw_landmarks = _raw_face_landmarks(face_image, known_face_locations, model="small")
  File "/usr/local/lib/python3.6/dist-packages/face_recognition/api.py", line 153, in _raw_face_landmarks
    face_locations = _raw_face_locations(face_image)
  File "/usr/local/lib/python3.6/dist-packages/face_recognition/api.py", line 102, in _raw_face_locations
    return face_detector(img, number_of_times_to_upsample)
TypeError: __call__(): incompatible function arguments. The following argument types are supported:
    1. (self: dlib.fhog_object_detector, image: array, upsample_num_times: int=0) -> dlib.rectangles

Invoked with: <dlib.fhog_object_detector object at 0x7f9ab0ea76f8>, None, 1

----

**Ubuntu 18.04 **:
**Howdy Latest **:

RogerBergling, 2018-09-26:
Nice work dude! Worked with device_id = 0
Thanx!

boltgolt, 2018-09-26:
In other cases like this, a reboot can break it again. Let me know if it's the case here.

closed 2018-09-26 by boltgolt

RogerBergling, 2018-09-26:
I will let you know ofcourse. But when a friend of mine try to login, it says, timeout and then get logged in... Is the a config that the face must be ok to get logged in....

boltgolt, 2018-09-26:
After the "timeout" message, Howdy triggers a `PAM_AUTH_ERR` response, which should **not** log you in. Are you using other auth modules? Does a timeout log you in when no one is behind the laptop?