boltgolt/howdy #80: Test not working

nilmerg, 2018-10-09:
Successfully installed howdy and got it working. Thanks!

Though, when I tried to test it (`sudo howdy test`) to see how it works I just got a grey-ish window and some errors on the cli.

```
Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x1e00003
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0x1e0000d
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x1e0000e
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x1e0000e
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x1ac
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0x1e0000f
X Error: BadPixmap (invalid Pixmap parameter) 4
  Major opcode: 54 (X_FreePixmap)
  Resource id:  0x1e0000e
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 2 (X_ShmDetach)
  Resource id:  0x1e0000d
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x1e00010
```

The one with opcode of 62 is then repeated infinetely once I interact with the window.

----

**Linux distribution (if applicable)**: Ubuntu 18.04
**Howdy version**: Latest from the ppa

nilmerg, 2018-10-09:
Oh, sorry, that's a duplicate of #60

closed 2018-10-09 by nilmerg

boltgolt, 2018-10-09:
Thanks for finding the right issue yourself!