boltgolt/howdy #81: Switching the camera

nilmerg, 2018-10-09:
During installation I've confirmed the first camera question that seemed to lit my camera. Though I had no idea which camera it was. (As I have no idea how it looks if my IR camera is active)

Running the test command I noticed the picture is full color. Not black-white.

Took then a look into the config. Noticed the camera is identified by id. Took a look into /dev and found `video0` as well as `video1`. Then set `1` as the camera in the configuration of howdy.

Though, running the test again gives me this:

```
VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV
VIDEOIO ERROR: V4L: can't open camera by index 1
```

Does this mean I cannot use my IR camera or did I miss something while changing the configuration?

The host I'm on is a Dell XPS 13 9370 (Developer Edition, Generic Ubuntu 18.04)

----

**Linux distribution (if applicable)**: Ubuntu 18.04
**Howdy version**: latest from ppa

boltgolt, 2018-10-09:
Could you open Cheese from the dash, go to Preferences and set your IR camera (video1) as the device? Does it show the right black and white image? Does changing resolution help anything?

nilmerg, 2018-10-10:
That's greyed out and only shows `Integrated_Webcam_HD: Integrate`.

It's a Dell. So I wouldn't be surprised if there's some driver missing or such.

Found then #10 and #6. As well as https://www.reddit.com/r/linux4noobs/comments/8u5z35/dell_xps_13_9370_ir_camera_drivers_for_linux/ which is from 3 months ago. The linked site for the patch is not reachable currently. I may also just try the oem kernel which was pre-installed.

Oh well, at least howdy works, even if not by use of the IR camera. Once I tried the patch or oem kernel I'll update this issue but I don't think that's anytime soon. Thanks for your time.

Ah, and for the record:

```
$ v4l2-ctl -l -d /dev/video0
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=0 value=0
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=1 max=8 step=1 default=1 value=1
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=10 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=3 value=3
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
              exposure_absolute 0x009a0902 (int)    : min=10 max=625 step=1 default=156 value=156 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
$ v4l2-ctl -l -d /dev/video1
```

boltgolt, 2018-10-15:
Thanks for looking into it! Your `v4l2-ctl` returns are exactly the same as for my (working) setup, so that's good. What definitely isn't good is that uvcvideo warning. There's not much i can do to fix it, but i'll add "Run `dmesg | grep "Unknown video format"`" to my list of troubleshooting steps.

pcvega, 2018-10-31:
Anch'io ho avuto gli stessi errori, non riesco a risolvere

boltgolt, 2018-10-31:
@pcvega please keep the conversation in English, i've translated your comment

pcvega, 2018-10-31:
I'm sorry, could I help me solve?

pcvega, 2018-10-31:
when I give the sudo howdy add command, it gives me this:

VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV
VIDEOIO ERROR: V4L: can't open camera by index 1

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 90, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 117, in <module>
    enc = face_recognition.face_encodings(frame)
  File "/usr/local/lib/python3.6/dist-packages/face_recognition/api.py", line 199, in face_encodings
    raw_landmarks = _raw_face_landmarks(face_image, known_face_locations, model="small")
  File "/usr/local/lib/python3.6/dist-packages/face_recognition/api.py", line 153, in _raw_face_landmarks
    face_locations = _raw_face_locations(face_image)
  File "/usr/local/lib/python3.6/dist-packages/face_recognition/api.py", line 102, in _raw_face_locations
    return face_detector(img, number_of_times_to_upsample)
TypeError: __call__(): incompatible function arguments. The following argument types are supported:
    1. (self: dlib.fhog_object_detector, image: array, upsample_num_times: int=0) -> dlib.rectangles

Invoked with: <dlib.fhog_object_detector object at 0x7fd7193913b0>, None, 1

boltgolt, 2018-11-01:
Could you run `dmesg | grep "Unknown video format"` and see if it outputs anything? There's a large chance that your camera is missing Linux drivers and will not work with Howdy.

sonic74, 2019-01-14:
I have the same problem, but the webcam in my XPS 13 9370 has the UVC 1.0 firmware:
`
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.551982] uvcvideo: Found UVC 1.00 device Integrated_Webcam_HD (0bda:58f4)
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.553909] uvcvideo 1-5:1.0: Entity type for entity Extension 4 was not initialized!
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.553914] uvcvideo 1-5:1.0: Entity type for entity Extension 7 was not initialized!
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.553916] uvcvideo 1-5:1.0: Entity type for entity Processing 2 was not initialized!
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.553917] uvcvideo 1-5:1.0: Entity type for entity Camera 1 was not initialized!
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.554001] input: Integrated_Webcam_HD: Integrate as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/input/input10
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.554595] uvcvideo: Unknown video format 00000032-0002-0010-8000-00aa00389b71
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.554600] uvcvideo: Found UVC 1.00 device Integrated_Webcam_HD (0bda:58f4)
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.556507] uvcvideo: Unable to create debugfs 1-2 directory.
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.556555] uvcvideo 1-5:1.2: Entity type for entity Extension 10 was not initialized!
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.556557] uvcvideo 1-5:1.2: Entity type for entity Extension 12 was not initialized!
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.556558] uvcvideo 1-5:1.2: Entity type for entity Processing 9 was not initialized!
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.556560] uvcvideo 1-5:1.2: Entity type for entity Camera 11 was not initialized!
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.556634] input: Integrated_Webcam_HD: Integrate as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.2/input/input11
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.556694] usbcore: registered new interface driver uvcvideo
Jan 10 12:11:56 sonic-XPS-13-9370 kernel: [    4.556695] USB Video Class driver (1.1.1)
`