boltgolt/howdy #82: IR camera stops working on Lenovo T480

XelNaha, 2018-10-10:
Hi,

I recently installed Howdy and I can see that the IR camera works with Howdy test, however it only works for 1 second, after which the IR led's turn off and essentially the recognition stops working.

----

Ubuntu 18.04
**Howdy version**:

boltgolt, 2018-10-15:
If you open Cheese Webcam Booth and set the device to your IR camera in preferences, does it suffer from the same issue? Are any errors printed to console?

XelNaha, 2018-10-16:
Hi Yes,

What I can see is that  the IR leds only flash once, and the image is clear. After which the LEDS no longer flash. I tried setting the settings to Jpeg which doesn't work.
I didn't see any errors in the console whilst doing this.

boltgolt, 2018-10-16:
Could you take a look at #51? Sometimes setting the resolution will fix this problem. You can try in out in Cheese in the preferences dialog.

kuilboer, 2018-10-24:
I have the same issue, ir leds flash once and then go dark, The situation is the same for Cheese. Without the IR LEDs the pickture is mostly green and does not provide enough detail to actually scan a face.

boltgolt, 2018-10-25:
The only fix for this specific issue so far is changing the default resolution. Open Cheese, got to ti Preferences and try all resolution options. Let me know if it changes anything.

boltgolt, 2018-11-09:
You can also try to set `force_mjpeg` to true in `sudo howdy config`

PancakeSparkle, 2018-11-09:
Cheese doesn't allow me to change resolution,realsense viewer says it's using 640x480,my ir camera image in cheese is fully green

boltgolt, 2018-11-09:
That's definitely a video decoding error, have you tried to set `force_mjpeg`?

PancakeSparkle, 2018-11-09:
Like this?
[image]

boltgolt, 2018-11-09:
Exactly like that

PancakeSparkle, 2018-11-09:
Still green for some reason,weird,because the intel software displays the IR image correctly

PancakeSparkle, 2018-11-09:
Even Intel's own software throws some warnings which might tell you something:
```pancakesparkle@pancakesparkle:~$ sudo realsense-viewer 
No face model known
[sudo] password for pancakesparkle: 
 10/11 00:44:26,576 WARNING [140276953159424] (backend-v4l2.cpp:1248) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
 10/11 00:44:26,576 WARNING [140276953159424] (backend-v4l2.cpp:1248) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
 10/11 00:44:26,577 WARNING [140276953159424] (sensor.cpp:338) Unregistered Media formats : [ INZC PAIR Y10  YUYV ]; Supported: [ ]
 10/11 00:44:26,853 ERROR [140277038934720] (tm-context.cpp:34) Failed to create TrackingManager
pancakesparkle@pancakesparkle:~$ 
```

wzrdtales, 2018-11-19:
@PancakeSparkle Check with GUVCview the available resolutions. Your resolution seems to be wrong, the t480 have the same IR cams installed as other series. And those have a res of 340x340

PancakeSparkle, 2018-11-19:
@wzrdtales I do not have a t480,i have an sr300,i have a desktop,the camera i'm using is usb connected, the dev just merged the issue i posted with this one because they're similar.

wzrdtales, 2018-11-19:
ah sry, my fault, still it looks pretty much like the bug I and the other of this issue had. And that was related to the resolution, may be checking out guvcview still helps. Cheese is not a good tool for dealing with settings for cams.

noreferences, 2018-12-29:
> The only fix for this specific issue so far is changing the default resolution. Open Cheese, got to to Preferences and try all resolution options. Let me know if it changes anything.

Managed to get rid of the green camera feed on ThinkPad T480s:

Setting resolution to 400x400 on Cheese made the green disappear. Then I saw a black and white picture of my face flashing, I turned off the lights in my room and I still could see my face, so I presumed that IR camera is working properly. 

For ```howdy test``` to display actual IR feed and not just a green screen, I had to change /lib/security/config.ini to frame_width = 400 and frame_height = 400 as well.

```howdy add``` works, authentication fails (I will update later, now I need to go to bed).

acegallagher, 2019-02-01:
I'm at the same state here as 'noreferences'. I have a t480s, set the resolution to 400x400 and get my face when I run test but then when howdy/PAM runs i get a 'no face model known' error from line 38 of pam.py. 

Some more information on the t480s camera:

https://blogs.gnome.org/jamesh/2017/10/22/thinkpad-infrared-camera/

boltgolt, 2019-02-02:
"no face model known" simply means that `sudo howdy add` has not been run yet for the current user, could you try adding a model?

acegallagher, 2019-02-02:
@boltgolt Sorry, I know my post was written hastily. I've added and cleared many models while debugging. I'm currently going through the code and trying to figure out why it thinks I might not have any. It's strange, because 'sudo howdy test' is able to find my face (red circle) but any PAM authentication request results in howdy claiming there's no models available. I've custom compiled several of the more recent 'releases' and the current git version.

boltgolt, 2019-02-02:
The circle in the test command only shows i detected _a_ face, not that i recognized it from a model. Does `sudo howdy list` show any models?

acegallagher, 2019-02-02:
It shows the four that I've currently added and am testing. 

https://i.imgur.com/anAPDra.png

boltgolt, 2019-02-02:
Are there any files in `/lib/security/howdy/models`?

boltgolt, 2019-02-03:
Id say you need to purge and reinstall at this point then, something is quite broken.

acegallagher, 2019-02-03:
@boltgolt 

Well. I had some time to figure it out. By adding 'pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, pamh.get_user()))' to the error output I was able to tell that pamh thought 'sudo' was being run by the root user. 

I have sudo configured to authenticate users by their access to the root account password via the config  lines:

> root ALL=(ALL) ALL
> Defaults targetpw  # Ask for the password of the target user
> ALL ALL=(ALL) ALL  # WARNING: only use this together with 'Defaults targetpw'

When I change this to simply authenticate via wheel/groups then pamh.get_user() returns my username. 

I'm not sure exactly how, or if, howdy can handle this but I'm sure it's the same problem that @noreferences was coming across. It's a fairly common sudo configuration. I figured I'd put it here for others to see

acegallagher, 2019-02-05:
@boltgolt 

When configured in the manner above it asks for root's password. 

I'm not sure either, just thought that the more information the better.

boltgolt, 2019-02-05:
You'll probably help some other folks if they're experiencing the same issue, so more information is definitely better. Thanks for taking the time to get to the bottom of this. I'm closing this issue for now because it now contains multiple different problems within the same thread.

closed 2019-02-05 by boltgolt

ju916, 2019-02-20:
I have a related problem. A realsense camera, that is connected via USB.to my systemm (Unbuntu 18.04).

+ it shows up as 2 devices (video0, video1). video0 is IR, video1 works in cheese as RGB, howdy can't use any of them

```
Bus 002 Device 005: ID 8086:0aa5 Intel Corp. 
  iManufacturer           1 Intel(R) RealSense(TM) Camera SR300
  iProduct                2 Intel(R) RealSense(TM) Camera SR300
```

Then I installed the realsense SDK (including special uvcvideo drivers). 
```
$ modinfo uvcvideo | grep "version:" 
version:        1.1.2.realsense-1.3.4
srcversion:     C4FB5E9B212E9C39ED6CD7C
```

Now I can see IR + RGB images/videos in realsense-viewer, but not in cheese, GUVCview or howdy. I am ready to provide additional info if needed.

btw: is there a list of verified working cameras?

boltgolt, 2019-02-20:
I'm locking this issue because it now contains some unrelated problems within the same thread.