boltgolt/howdy #83: File permissions in PPA package

GJDitchfield, 2018-10-13:
I installed Howdy 2.3.1 from the PPA on an 18.04 Kubuntu system.  Most authentication worked, but the Plasma desktop's lock screen wouldn't even try face recognition.

auth.log contained "/lib/security/howdy/pam.py[_n_]: Can not open module: Permission denied", so I applied the fix from issue #28.  If the package could set the permissions when it installs the files, that would be great.

The file permissions I ended up with are 
```
 drwxr-xr-x   3 root     root         4096 Oct 11 21:05 /lib/security/
 drwxr-xr-x   4 root     root         4096 Oct 11 21:11 /lib/security/howdy
 -rwxr-xr-x   1 root     root         2599 Sep 24 11:46 /lib/security/howdy/pam.py
 drwxr-xr-x   3 root     root         4096 Oct 11 21:11 /lib/security/howdy/cli
 -rwxr-xr-x   1 root     root         3997 Sep 24 11:46 /lib/security/howdy/cli/add.py
 drwxr-xr-x   2 root     root         4096 Oct 11 21:11 /lib/security/howdy/cli/__pycache__
 -rw-r--r--   1 root     root         2370 Oct 11 21:11 /lib/security/howdy/cli/__pycache__/add.cpython-36.pyc
 -rw-r--r--   1 root     root          118 Oct 11 21:11 /lib/security/howdy/cli/__pycache__/__init__.cpython-36.pyc
 -rwxr-xr-x   1 root     root         2432 Apr 13  2018 /lib/security/howdy/cli/remove.py
 -rwxr-xr-x   1 root     root         1474 May 11 03:36 /lib/security/howdy/cli/disable.py
 -rwxr-xr-x   1 root     root          340 Sep 24 10:13 /lib/security/howdy/cli/config.py
 -rwxr-xr-x   1 root     root           34 Apr 13  2018 /lib/security/howdy/cli/__init__.py
 -rwxr-xr-x   1 root     root         5162 Sep 24 10:13 /lib/security/howdy/cli/test.py
 -rwxr-xr-x   1 root     root         1361 Apr 13  2018 /lib/security/howdy/cli/list.py
 -rwxr-xr-x   1 root     root         1004 Apr 13  2018 /lib/security/howdy/cli/clear.py
 -rwxr-xr-x   1 root     root         3314 Sep 24 10:13 /lib/security/howdy/cli.py
 drwxr-xr-x   2 root     root         4096 Oct 11 21:12 /lib/security/howdy/models
 -rw-r--r--   1 root     root         2859 Oct 11 21:12 /lib/security/howdy/models/gjditchf.dat
 -rw-r--r--   1 root     root         1698 Oct 11 21:08 /lib/security/howdy/config.ini
 -rwxr-xr-x   1 root     root         5413 Sep 24 10:40 /lib/security/howdy/compare.py
```

boltgolt, 2018-10-13:
The problem is that anyone could edit `compare.py` if it's owned by the user. Any userspace process could just add a `return True` and authenticate without user input. For now people with this problem will need to use this insecure fix, but there probably is a better solution.

boltgolt, 2018-10-15:
First of all i have to apologize, i misread your first comment in it's entirety. I thought you meant setting the owner as the user and i'm not sure why. 

As for the default permissions: I would like to give as few permissions as possible, but in this case i can't really think of a way execute on `compare.py` could be exploited. Do other files need explicit execute?

GJDitchfield, 2018-10-15:
I thought the execute permissions were necessary, but I must have gotten confused during my experiments.  I removed some permissions, and this still works for me:
```
 drwxr-xr-x   4 root     root         4096 Oct 11 21:11 /lib/security/howdy/
 -rw----r--   1 root     root         2599 Sep 24 11:46 /lib/security/howdy/pam.py
 drwx---r--   3 root     root         4096 Oct 11 21:11 /lib/security/howdy/cli
 -rwxr--r--   1 root     root         3997 Sep 24 11:46 /lib/security/howdy/cli/add.py
 drwxr-xr-x   2 root     root         4096 Oct 11 21:11 /lib/security/howdy/cli/__pycache__
 -rw-r--r--   1 root     root         2370 Oct 11 21:11 /lib/security/howdy/cli/__pycache__/add.cpython-36.pyc
 -rw-r--r--   1 root     root          118 Oct 11 21:11 /lib/security/howdy/cli/__pycache__/__init__.cpython-36.pyc
 -rwxr--r--   1 root     root         2432 Apr 13  2018 /lib/security/howdy/cli/remove.py
 -rwxr--r--   1 root     root         1474 May 11 03:36 /lib/security/howdy/cli/disable.py
 -rwxr--r--   1 root     root          340 Sep 24 10:13 /lib/security/howdy/cli/config.py
 -rwxr--r--   1 root     root           34 Apr 13  2018 /lib/security/howdy/cli/__init__.py
 -rwxr--r--   1 root     root         5162 Sep 24 10:13 /lib/security/howdy/cli/test.py
 -rwxr--r--   1 root     root         1361 Apr 13  2018 /lib/security/howdy/cli/list.py
 -rwxr--r--   1 root     root         1004 Apr 13  2018 /lib/security/howdy/cli/clear.py
 -rwxr-xr-x   1 root     root         3314 Sep 24 10:13 /lib/security/howdy/cli.py
 drwxr-xr-x   2 root     root         4096 Oct 11 21:12 /lib/security/howdy/models
 -rw-r--r--   1 root     root         2859 Oct 11 21:12 /lib/security/howdy/models/gjditchf.dat
 -rw----r--   1 root     root         1698 Oct 11 21:08 /lib/security/howdy/config.ini
 -rw----r--   1 root     root         5413 Sep 24 10:40 /lib/security/howdy/compare.py
```

boltgolt, 2018-10-16:
I'll update the permissions to that in the next release, thanks!

closed 2018-11-09 by boltgolt

MK73DS, 2023-05-17:
I don't know if it's okay for me to post in this closed issue, but I'm getting `Can not open module: Permission denied` error when I want to login after sleep (or manually locking my session). This is probably since howdy is executed as me, not root, which may be an issue with my display manager (sddm) or desktop environment (kde plasma).

If I set the permissions like posted in this issue thread, it doesn't work, I still get the permission denied error.

I found that setting the whole `/lib/security/howdy` folder to `755` (`rwxr-xr-x`) works, but I'd like to know if this can be a security issue or not. I guess the only problem is giving write permission to any user, which I'm not doing, but other than that I don't know how much of a security issue that it. Maybe having read access to the `models` folder is? But howdy won't work after sleep if I don't give it read permission... 

Thanks in advance. I'm using Arch Linux with Kde Plasma.