boltgolt/howdy #87: test fail

udipta, 2018-10-23:
The test window remains blank and does work. [
[image]
](url)
$sudo howdy test

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x4600003
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0x460000d
[image]

boltgolt, 2018-10-23:
Please use the search function before opening a new issue. An open issue with this exact error and a solution can be found at #60. Your error is even in the title.

closed 2018-10-23 by boltgolt

udipta, 2018-10-23:
I'll keep in mind next time and thanks for the solution