boltgolt/howdy #88: Alienware 15R3 IR camera Drivers

ntej, 2018-10-26:
_Please describe the issue in as much detail as possible, including any errors and traces_

Howdy is working perfectly with the Regular camera but I can't use IR camera as it is not detected by Ubuntu 18.04 beacuse of drivers issue i guess. 
```bash
$ls -ltrh /dev/video*
crw-rw----+ 1 root video 81, 0 Oct 26 02:02 /dev/video0
```
I searched for hours but no luck. Please help me finding the IR camera drivers or any working technique so that i can use the Howdy with IR Camera.

boltgolt, 2018-10-26:
That does sound like a driver issue, does running `dmesg | grep "Unknown video format"` output anything?

boltgolt, 2018-11-06:
You could take a manual look at `dmesg` and search for video related errors, especially those mentioning `uvcvideo`. Does Cheese show the IR camera in Preferences?

sapjunior, 2018-11-15:
Try to update to kernel 4.19. My archlinux XPS13 9370 IR camera is working after update the kernel

ntej, 2018-11-15:
I didn't get time to look more into the issue. @boltgolt cheese is showing only the regular camera in preferences. I will look more into the dmesg logs and get back. 

@sapjuniorI will give a try and see, thanks!

boltgolt, 2018-11-19:
@sapjunior what's your exact kernel version number?

sapjunior, 2018-11-20:
I'm using 4.19.2-1-ck-skylake

ntej, 2018-12-06:
@boltgolt I looked into DMESG logs and grep with the camera I found following lines

` 18.698893] uvcvideo: Found UVC 1.00 device Integrated_Webcam_HD (0bda:58c2)
[   18.706178] uvcvideo 1-7:1.0: Entity type for entity Extension 4 was not initialized!
[   18.706179] uvcvideo 1-7:1.0: Entity type for entity Extension 7 was not initialized!
[   18.706180] uvcvideo 1-7:1.0: Entity type for entity Processing 2 was not initialized!
[   18.706181] uvcvideo 1-7:1.0: Entity type for entity Camera 1 was not initialized!
[   18.706225] input: Integrated_Webcam_HD: Integrate as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/input/input19
[   18.706262] usbcore: registered new interface driver uvcvideo
[   18.706262] USB Video Class driver (1.1.1)`

I see some 'camera 1' was not initialized which i think is IR camera ?

boltgolt, 2018-12-06:
There's a [similar bug](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1626731) on launchpad, could you try running `sudo apt install uvcdynctrl` and rebooting?

ntrepid8, 2019-01-25:
I noticed this issue on a recent Dell XPS 13 9370. It looks like there was a is video format for 8-bit IR:
```
IR 8-bit (L8_IR)
```

It only made it into the kernel for the first time in 4.19:
- https://patchwork.kernel.org/patch/10298995/
- https://github.com/torvalds/linux/commit/557a5c7fe6503230f6a3a41441981aed6e897d17

MooCowsAreAwesome, 2019-01-25:
Thanks for follow up on this!  So it should work once I update the kernel then?

bkmagnetron, 2019-07-13:
IR camera is still not working.
HW: Alienware 15R3
Distro: Ubuntu Studio 19.04
Kernel: 5.0.0-20-lowlatency

shahbazshueb, 2019-08-07:
> IR camera is still not working but the webcam works.
> HW: Alienware 15R3
> Distro: Ubuntu Studio 19.04
> Kernel: 5.0.0-20-lowlatency
> 
> ```
> $ ls -ltrh /dev/video*
> crw-rw----+ 1 root video 81, 1 Jul 14 01:27 /dev/video1
> crw-rw----+ 1 root video 81, 0 Jul 14 01:27 /dev/video0
> ```

HW: Alienware 13R3
Not working for me too :(

phckopper, 2020-02-03:
Dell Inspiron 13 5378 not working.

dmesg:
`[    6.126056] uvcvideo: Found UVC 1.00 device Integrated_Webcam_HD (0bda:58c2)
[    6.165605] uvcvideo 1-5:1.0: Entity type for entity Extension 4 was not initialized!
[    6.165608] uvcvideo 1-5:1.0: Entity type for entity Extension 7 was not initialized!
[    6.165612] uvcvideo 1-5:1.0: Entity type for entity Processing 2 was not initialized!
[    6.165614] uvcvideo 1-5:1.0: Entity type for entity Camera 1 was not initialized!
[    6.166943] usbcore: registered new interface driver uvcvideo`

lsusb:
`Bus 001 Device 002: ID 0bda:58c2 Realtek Semiconductor Corp. Integrated_Webcam_HD`

emilygeil, 2021-02-26:
Similar issue on Ubuntu 20.04, kernel 5.8.0, Dell Inspiron 13 5378 -- I'm sure this is a driver problem rather than a howdy problem, but I figured I'd update the information on this

> That does sound like a driver issue, does running `dmesg | grep "Unknown video format"` output anything?

No output

> You could take a manual look at `dmesg` and search for video related errors, especially those mentioning `uvcvideo`. Does Cheese show the IR camera in Preferences?

I looked through but didn't notice any -- the only lines mentioning uvc were:
```
[    3.772662] videodev: Linux video capture interface: v2.00
[    3.802933] uvcvideo: Found UVC 1.00 device Integrated_Webcam_HD (0bda:58c6)
[    3.829097] input: Integrated_Webcam_HD: Integrate as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/input/input20
[    3.829685] usbcore: registered new interface driver uvcvideo
[    3.829686] USB Video Class driver (1.1.1)
```

> There's a [similar bug](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1626731) on launchpad, could you try running `sudo apt install uvcdynctrl` and rebooting?

Unfortunately, no effect

Output of `v4l2-ctl -d /dev/v4l/by-id/usb-CNFGE11F301030046600_Integrated_Webcam_HD_0x001-video-index1 --list-formats-ext`
```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```

Output of `ffmpeg -f v4l2 -list_formats all -i /dev/video1`
```
ffmpeg version 4.2.4-1ubuntu0.1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.3.0-10ubuntu2)
  configuration: --prefix=/usr --extra-version=1ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-nvenc --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
[video4linux2,v4l2 @ 0x556aa2b876c0] ioctl(VIDIOC_G_INPUT): Inappropriate ioctl for device
/dev/video1: Inappropriate ioctl for device
```

The normal webcam is on `/dev/video0` and has no issue -- running the above commands on it has the expected output

boltgolt, 2022-02-09:
Closing as stale and out of scope, driver issue

closed 2022-02-09 by boltgolt