boltgolt/howdy #89: /etc/alternatives notion doesn't exist in Arch, or anything besides Ubuntu/Debian

yassineim, 2018-11-01:
howdy config should probably use something like $EDITOR

boltgolt, 2018-11-02:
`$EDITOR` doesn't exist in Ubuntu though, i'm open for any suggestions that work on all distros

yassineim, 2018-11-02:
/bin/nano ?

<https://github.com/boltgolt/howdy/issues/89#issuecomment-435364571>, or mute
the thread

boltgolt, 2018-11-03:
That would hardcode nano as editor, even for Ubuntu users that have a different default. I'll add a check for `/etc/alternatives/editor` and run `/bin/nano` if it doesn't exist.

KelleyMcChes, 2018-11-03:
Would it bee better to check whether the program is running on Arch or Ubuntu and then use /etc/alternatives or $EDITOR respectively. That way, people who either don't like nano or don't have nano installed can use it

boltgolt, 2018-11-03:
Yeah, good point. I'll keep nano as fallback so it'll go: Try `/etc/alternatives` > try `$EDITOR` > run `/bin/nano/`.

closed 2018-11-09 by boltgolt