boltgolt/howdy #90: Howdy always uses activate Python interpreter

EdwardJB, 2018-11-03:
I've got several virtual environments for Python development. Howdy will always use the active one, which most the time isn't the one with its dependencies. So I get errors like this:

```
ed@pop:~/Repositories/mariano$ source activate scratch
(scratch) ed@pop:~/Repositories/mariano$ which python
/home/ed/miniconda3/envs/scratch/bin/python
(scratch) ed@pop:~/Repositories/mariano$ sudo ls
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 11, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
Unknown error: 1
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 11, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
Unknown error: 1
[sudo] password for ed:
```
If I specifically activate the right environment, it works - but obviously that's a nuisance:

```
(scratch) ed@pop:~/Repositories/mariano$ source activate
(base) ed@pop:~/Repositories/mariano$ which python
/home/ed/miniconda3/bin/python
(base) ed@pop:~/Repositories/mariano$ sudo ls
Identified face as ed
test.c
```

Using the activate Python makes sense in most cases, but not when doing system-wide authentication.

boltgolt, 2018-11-06:
I'll change the command to directly run `/usr/bin/python3`, which should get around the enviroment $PATH

closed 2018-11-09 by boltgolt

astanziola, 2021-07-09:
I think this is possibly still an issue. 
Here is from a `base` conda environment in Arch, after a fresh howdy install

```zsh
sudo howdy add                                                                ✔  3m 8s   base  
[sudo] password for antonio: 
Traceback (most recent call last):
  File "/usr/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 11, in <module>
    from recorders.video_capture import VideoCapture
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 6, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
```

boltgolt, 2021-08-12:
That's probably an incorrect opencv instation, Howdy not hardlinks to python3