boltgolt/howdy #92: Keyerror: Video

rmconway, 2018-11-06:
Whenever I do something that requires a password, Howdy tries to run, but I get this error.

`Traceback (most recent call last):
  File "/usr/lib/python3.6/configparser.py", line 1138, in _unify_values
    sectiondict = self._sections[section]
KeyError: 'video'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 61, in <module>
  File "/usr/lib/python3.6/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.6/configparser.py", line 1141, in _unify_values
    raise NoSectionError(section)
configparser.NoSectionError: No section: 'video'
Unknown error: 1
Traceback (most recent call last):
  File "/usr/lib/python3.6/configparser.py", line 1138, in _unify_values
    sectiondict = self._sections[section]
KeyError: 'video'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 61, in <module>
  File "/usr/lib/python3.6/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.6/configparser.py", line 1141, in _unify_values
    raise NoSectionError(section)
configparser.NoSectionError: No section: 'video'
Unknown error: 1
`

----

- [ ] I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 18.10

Howdy version: 2.3.1

boltgolt, 2018-11-06:
Could you post the contents of `sudo howdy config`?

rmconway, 2018-11-06:
# Howdy config file
# Press CTRL + X to save in the nano editor

[core]
# Do not print anything when a face verification succeeds
no_confirmation = false

# When a user without a known face model tries to use this script, don't
# show an error but fail silently
suppress_unknown = false

# Disable Howdy in remote shells
ignore_ssh = true

# Auto dismiss lock screen on confirmation
# Will run loginctl unlock-sessions after every auth
# Expirimental, can behave incorrectly on some systems
dismiss_lockscreen = false

# Disable howdy in the PAM
# The howdy command will still function
disabled = false

[video]
# The certainty of the detected face belonging to the user of the account
# On a scale from 1 to 10, values above 5 are not recommended
certainty = 3.5

# The number of seconds to search before timing out
timout = 4

# The /dev/videoX id to capture frames from
# Should be set automatically by the installer
device_id = 0

# Scale down the video feed to this maximum height
# Speeds up face recognition but can make it less precise
max_height = 320

# Force the use of Motion JPEG when decoding frames, fixes issues with
# YUYV raw frame deconding
force_mjpeg = false

# Set the camera input profile to this width and height
# The largest profile will be used if set to -1
# Automatically ignored if not a valid profile
frame_width = -1
frame_height = -1
# Because of flashing IR emitters, some frames can be completely unlit
# Skip the frame if the lowest 1/8 of the histogram is above this percentage
# of the total
# The lower this setting is, the more dark frames are ignored
dark_threshold = 50

[debug]
# Show a short but detailed diagnostic report in console
end_report = false

rmconway, 2018-11-06:
Also, my distro is Ubuntu 18.10, not Debian. Thanks for the quick response.

boltgolt, 2018-11-19:
Looks good, which means Howdy can't find the file itself. Did you move config.ini or change it's permissions?

dmig, 2018-12-09:
What are these `\#` at the beginning of each line? Likely these slashes break config parsing.

boltgolt, 2018-12-09:
They are escaped markdown formatting from before i edited his reply to put his config in a codeblock for better readability. Fixed now.

boltgolt, 2020-01-07:
No author response for over a year, closing as stale

closed 2020-01-07 by boltgolt