boltgolt/howdy #93: Camera green in howdy test and cheese

PancakeSparkle, 2018-11-09:
Hey,
Trying to add a face to howdy gives me a "No face detected, aborting" message, camera is a Razer Stargazer, even tried the Realsense viewer and it works there, it also shows a green screen in cheese, and I cannot change the resolution from 640x480, RGB camera works,OS is ubuntu 18.10,tried enabling mjpeg in config file and changing the frame width to no avail,still green.
Thanks.

boltgolt, 2018-11-09:
Hey, thanks for reporting this. It seems like you have about the same problems as the 2 people from issue #82, so to keep it all together i'm closing this.

closed 2018-11-09 by boltgolt

PancakeSparkle, 2018-11-09:
The LEDs don't turn off on mine,they're solid,and the image is full green