boltgolt/howdy #95: Python errors

rusins, 2018-11-11:
Getting different python errors depending on what command I run.
```
$ sudo howdy config
Opening config.ini in the default editor
Traceback (most recent call last):
  File "/usr/bin/howdy", line 94, in <module>
    import cli.config
  File "/usr/lib/security/howdy/cli/config.py", line 12, in <module>
    subprocess.call(["/etc/alternatives/editor", os.path.dirname(os.path.realpath(__file__))  + "/../config.ini"])
  File "/usr/lib/python3.7/subprocess.py", line 317, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.7/subprocess.py", line 769, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1516, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/etc/alternatives/editor': '/etc/alternatives/editor'
```
Alright, can't figure out my editor because Arch doesn't use update-alternatives. (I guess?)

```
$ sudo howdy add
Adding face model for the user raitis
Enter a label for this new model [Initial model]: 

(python3:6919): GStreamer-CRITICAL **: 01:10:36.534: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed
Unable to stop the stream: Invalid argument
VIDEOIO ERROR: V4L: can't open camera by index 1

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/bin/howdy", line 90, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 117, in <module>
    enc = face_recognition.face_encodings(frame)
  File "/usr/lib/python3.7/site-packages/face_recognition/api.py", line 209, in face_encodings
    raw_landmarks = _raw_face_landmarks(face_image, known_face_locations, model="small")
  File "/usr/lib/python3.7/site-packages/face_recognition/api.py", line 153, in _raw_face_landmarks
    face_locations = _raw_face_locations(face_image)
  File "/usr/lib/python3.7/site-packages/face_recognition/api.py", line 102, in _raw_face_locations
    return face_detector(img, number_of_times_to_upsample)
TypeError: __call__(): incompatible function arguments. The following argument types are supported:
    1. (self: dlib.fhog_object_detector, image: array, upsample_num_times: int=0) -> dlib.rectangles

Invoked with: <dlib.fhog_object_detector object at 0x7f8c779ce298>, None, 1
```

```
$ sudo howdy test

(python3:6968): GStreamer-CRITICAL **: 01:11:42.343: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed
Unable to stop the stream: Invalid argument
VIDEOIO ERROR: V4L: can't open camera by index 1

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Traceback (most recent call last):
  File "/usr/bin/howdy", line 102, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 85, in <module>
    overlay = frame.copy()
AttributeError: 'NoneType' object has no attribute 'copy'
```
----

I've searched for similar issues already, and my issue has not been reported yet. (I have a track-record for sucking at searching for existing issues though :disappointed: )

Linux distribution: Arch linux

Howdy version: 2.3.1-1

boltgolt, 2018-11-12:
So there are 2 issues here:

The first problem happens on all Arch installations because i did not realize `/etc/alternatives/editor` did not exist on there. It is fixed in 2.4.0 but that version does not have an Arch release yet.

Both the second and third errors stem from the same problem: You're trying to use `/dev/video1` as a camera input but it's not readable. It could be that you just need to change `device_id` in your config to the right number, but unfortunately it's much more likely that your camera is not supported by linux drivers. Try using other programs (like Cheese on Ubuntu) to go though your available cameras and look for the back and white IR input.

rusins, 2018-11-12:
About the camera 
[image]

rusins, 2018-11-12:
Accidentally closed, whoops.

I appear to have 4 video devices in /dev/, and cheese works with all of them.

Another thing I thought I'd mention: with howdy failing like this at the moment, when I tried to unlock my Gnome session, it wouldn't allow me to bring up the password field to type. Uninstalling Howdy fixed it. Not sure if an issue per se, but still silly that it prevents me from logging in normally.

boltgolt, 2018-11-12:
You can go to Preferences in Cheese to select a different video input. I'd also suggest playing with the frame size, as some cameras only seem to work on certain resolutions.

Howdy should drop authentication to the next module on error. On most systems this means it'll drop to unix password auth. Very strange that it failed to do so in your case. Does `/var/log/auth.log` show any errors?

Aster-the-Med-Stu, 2019-01-06:
@boltgolt Same issue here under Manjaro. The problem is that under this scenario it is impossible to add a face, letting alone use it to auth...

mdnahas, 2019-02-12:
This may be the same as this issue?
[https://github.com/boltgolt/howdy/issues/70](https://github.com/boltgolt/howdy/issues/70)

Tip of the hat to https://aur.archlinux.org/packages/howdy/

I've got the same issue with "sudo howdy add" (see below).  

Running on Dell XPS 13 using Arch (Antergos 19.1).  "sudo howdy test" seems to work (with /dev/video2), I get a black-and-white image and the IR illuminating LEDs light up. A black circle blinks around my face.  Running version 0.3.0 of AUR package "python-face_recognition_models" and version 2.3.1 of AUR package "howdy".

I haven't fixed the problem yet.  I'll either try @sapjunior 's solution in issue 70 or see if I can upgrade to a newer version of Howdy.

```
  File "/usr/lib/python3.7/site-packages/face_recognition/api.py", line 210, in <listcomp>
    return [np.array(face_encoder.compute_face_descriptor(face_image, raw_landmark_set, num_jitters)) for raw_landmark_set in raw_landmarks]
TypeError: compute_face_descriptor(): incompatible function arguments. The following argument types are supported:
    1. (self: dlib.face_recognition_model_v1, img: numpy.ndarray[(rows,cols,3),uint8], face: dlib.full_object_detection, num_jitters: int=0, padding: float=0.25) -> dlib.vector
    2. (self: dlib.face_recognition_model_v1, img: numpy.ndarray[(rows,cols,3),uint8], faces: dlib.full_object_detections, num_jitters: int=0, padding: float=0.25) -> dlib.vectors
    3. (self: dlib.face_recognition_model_v1, batch_img: List[numpy.ndarray[(rows,cols,3),uint8]], batch_faces: List[dlib.full_object_detections], num_jitters: int=0, padding: float=0.25) -> dlib.vectorss

Invoked with: <dlib.face_recognition_model_v1 object at 0x7f3aabdad7d8>, array([[ 51,  51,  49, ...,  55,  52,  54],
       [ 51,  51,  51, ...,  52,  51,  52],
       [ 50,  53,  50, ...,  55,  53,  54],
       ...,
       [255, 255, 255, ..., 255, 255, 255],
       [255, 255, 255, ..., 255, 255, 255],
       [255, 255, 255, ..., 255, 255, 255]], dtype=uint8), <dlib.full_object_detection object at 0x7f3aabdad6f8>, 1
```

TheOneHyer, 2019-02-21:
I am running Manjaro and have exactly the same issue as @mdnahas.

TheOneHyer, 2019-02-21:
I should be more useful in my comment than just saying I have an issue too. I am willing to provide any help the developers need to get this resolved. It may help to note that `sudo howdy test` results in a window popping up showing my face real-time in black and white (which sounds like Issue #70, iirc) but `howdy add` results in the same error as above.

boltgolt, 2020-01-07:
If it was similar to #70, a fix has been implemented. Either way, closing due to inactivity

closed 2020-01-07 by boltgolt