boltgolt/howdy #96: I'd love to see the ability to use two cameras

JoeRomeo, 2018-11-17:
At the office, I have an external USB camera and on the go the laptop has one built in. I don't know how I'd get this working on both.

dmig, 2018-11-18:
Just edit app config, specify an external device. But you'll probably need to add more face samples...

I'd propose a small patch to support camera fallback:
```diff
--- compare.py.original	2018-11-18 12:29:05.533869737 +0700
+++ compare.py	2018-11-18 12:29:40.258341917 +0700
@@ -58,7 +58,14 @@
 timings.append(time.time())
 
 # Start video capture on the IR camera
-video_capture = cv2.VideoCapture(config.get("video", "device_path"))
+devices = (d.strip() for d in config.get("video", "device_path").split(','))
+for dev in devices:
+    if not os.path.exists(dev):
+        continue
+    video_capture = cv2.VideoCapture(dev)
+    break
+else:
+    sys.exit(1) # not sure about exit code
 
 # Force MJPEG decoding if true
 if config.get("video", "force_mjpeg") == "true":
```

This patch will allow to specify multiple device paths and pick up first available camera.

Or did you mean both cams simultaneously (like windoze does on my laptop)?

boltgolt, 2018-11-19:
I appreciate the effort, but i think this might be out of scope. I just don't see many uses for this feature and more code to maintain.

However, you could solve this with a dynamic symlink. Make a dummy symlink and configure Howdy to use it as device path. A very simple python script could then point that symlink to the right camera for the situation.

closed 2018-11-19 by boltgolt