boltgolt/howdy #100: OpenCV Upgrade results in error: recursion is detected during loading of "cv2"

wzrdtales, 2018-11-19:
Hey there,

I just upgraded opencv (upgraded opencv (3.4.3-4 -> 3.4.3-5)) and this breaks howdy for some reason. I downgraded again to the previous version and everything starts working again. Haven't had time to dig deeper, but this is the error:

```
['/build/opencv/src/build/lib/python3', '/usr/lib/security/howdy', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/usr/lib/python3.7/site-packages']
Traceback (most recent call last):
  File "/usr/sbin/howdy", line 102, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 5, in <module>
    import cv2
  File "/usr/lib/python3.7/site-packages/cv2/__init__.py", line 89, in <module>
    bootstrap()
  File "/usr/lib/python3.7/site-packages/cv2/__init__.py", line 79, in bootstrap
    import cv2
  File "/usr/lib/python3.7/site-packages/cv2/__init__.py", line 89, in <module>
    bootstrap()
  File "/usr/lib/python3.7/site-packages/cv2/__init__.py", line 23, in bootstrap
    raise ImportError('ERROR: recursion is detected during loading of "cv2" binary extensions. Check OpenCV installation.')
ImportError: ERROR: recursion is detected during loading of "cv2" binary extensions. Check OpenCV installation.
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): arch

Howdy version: latest

boltgolt, 2018-11-21:
Looks like an arch [bug report](https://bugs.archlinux.org/task/60855) has been made already and a fix is available on the [opencv repo](https://github.com/opencv/opencv/issues/13202). Very curious en might be impacting everyone on arch soon.

wangqr, 2018-11-21:
Is the `OPENCV_SKIP_PYTHON_LOADER=ON` solution a recommended solution to this issue or just a temporary workaround?

lightonflux, 2020-06-03:
@wzrdtales can you confirm that the issue is fixed?

musikid, 2023-10-18:
Feel free to reopen this if you don't think that it's solved.

closed 2023-10-18 by musikid