boltgolt/howdy #101: howdy add dont work

ryancotty18, 2018-11-26:
_Please describe the issue in as much detail as possible, including any errors and traces_

hello trying to setup howdy on my linux but when i do sudo howdy add  im getting Illegal instruction was wounding if u can help me please 
thanks ryan

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): kde neon

Howdy version:

boltgolt, 2018-11-26:
Did you install Howdy through the PPA? Is "Illegal instruction" the only error returned?

ryancotty18, 2018-11-26:
i followed the website page https://github.com/boltgolt/howdy to install i never got no errors not that i no off so after installed i wanted to add face model so i put sudo howdy add then illegal instruction came up

dmig, 2018-12-09:
Please wait for the next release, will be fixed there. Looks like your CPU isn't capable of AVX instructions.

dmig, 2018-12-09:
Right now you can fix it manually (and gain a bit more performance):
1. first check what your CPU is capable of:
```shell
grep -Em1 --color=always 'sse|avx' /proc/cpuinfo
```
And choose first available from: AVX, SSE4, SSE3, SSE2. If none available, remove `--yes USE_xxx_INSTRUVTIONS` flag from command below.

2. then grab `dlib` and rebuild it yourself:
```shell
git clone https://github.com/davisking/dlib dlib

sudo apt install libatlas3-base libatlas-base-dev
cd dlib
# replace xxx with one from the first step: AVX, SSE4, SSE3, SSE2. Case matters!
sudo python3 setup.py install --yes USE_xxx_INSTRUCTIONS
cd ..
sudo apt remove libatlas-base-dev
sudo rm -rf dlib
```
You're done.

boltgolt, 2019-01-07:
Should be fixed with the 2.5.0 release, closing

closed 2019-01-07 by boltgolt

Nos78, 2022-11-30:
> 
```shell
[noscere@meg ~]$ grep -Em1 --color=always 'sse|avx' /proc/cpuinfo
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer rdrand lahf_lm 3dnowprefetch epb pti ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat md_clear
```

Nos78, 2022-12-02:
Is it worth me opening a new issue for this, although it looks like the same issue; this ticket is marked as closed, and it relates to a debian distro rather than Manjaro/AUR which I am using. Anyway, additional to my previous post, I tried `sudo howdy -U noscere snapshot` and got the following:

```shell
    [noscere@meg ~]$ sudo howdy -U noscere snapshot
    [ WARN:0@0.486] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (2401)
    handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: 
    Could not read from resource.
    [ WARN:0@0.487] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (1356) open 
    OpenCV | GStreamer warning: unable to start pipeline
    [ WARN:0@0.487] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (862) 
    isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
    Generated snapshot saved as
    /usr/lib/security/howdy/snapshots/20221202T234942.jpg
    [noscere@meg ~]$ sudo howdy -U noscere add
    Illegal instruction
```

So snapshots work, but cannot use the add command. From the config, I manually configured my device (using the working config from my ubuntu laptop installation of howdy):

```bash
    # The path of the device to capture frames from
    # Should be set automatically by an installer if your distro has one
    device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:3:1.0-video-index0
```

```bash
[noscere@meg ~]$ ls /dev/v4l/by-path/
pci-0000:00:14.0-usb-0:3:1.0-video-index0  pci-0000:00:14.0-usb-0:3:1.0-video-index1
```

```bash
    [noscere@meg ~]$ v4l2-ctl --list-devices
    Lenovo EasyCamera: Lenovo EasyC (usb-0000:00:14.0-3):
            /dev/video0
            /dev/video1
            /dev/media0
```
And for further information, to show I am using the correct device:
```shell
    [noscere@meg ~]$ v4l2-ctl -d /dev/video0 --all
    Driver Info:
            Driver name      : uvcvideo
            Card type        : Lenovo EasyCamera: Lenovo EasyC
            Bus info         : usb-0000:00:14.0-3
            Driver version   : 5.15.78
            Capabilities     : 0x84a00001
                    Video Capture
                    Metadata Capture
                    Streaming
                    Extended Pix Format
                    Device Capabilities
```