boltgolt/howdy #102: Running howdy on RHEL (7.5)

williamwlk, 2018-11-27:
Hi,

Saw a friend using howdy on his Ubuntu Desktop and I became jealous and curious if I could make it work on my Red Hat Desktop.

After having destroyed two vms and by trial and error (for two weeks), I have finally made howdy work perfectly on my RHEL7.5 VM. :100: 

Will do a write up soon.

Have seen/read the [fedora issue](https://github.com/boltgolt/howdy/issues/26) but RHEL is in principal a different conservative enterprise distro.

I guess I can make it work on CentOS as well. And I will check if I could help with the Fedora Folks.

Great work @boltgolt Great System Integration Work and Great App. :+1: 

Regards,
Will

boltgolt, 2018-11-28:
Nice work! As Fedora and Redhat still use the same packaging process, would that mean that your work would also run on Fedora? Would there be a big difference in installation instructions between CentOS, Fedora and RHEL?

luyatshimbalanga, 2018-11-28:
How about using the spec file I made as template and add RHEL/CentOS specific conditional macros [from the guideline](https://rpm-packaging-guide.github.io/#more-on-macros)? That method will greatly reduce unnecessary duplications.

williamwlk, 2018-11-28:
Hi @boltgolt Hi @luyatshimbalanga 

A quick reply : Merging is a possibility coz my rpm work is 0% done. I made everything work from scratch and sources. 

howdy main app is a quite straight forward one to be packaged. But the dependencies are not.

My writeup is coming in hot shortly for further discussion. Thank you.

Regards,
Will

williamwlk, 2018-11-28:
Hi @boltgolt Hi @luyatshimbalanga 

Here we go the writeup. 

MY THOUGHTS:
- Don't mind the "I made it work" silly or otherwise omitted steps.
- dependencies of dependencies we should pay attention to.
- python2 and python3 need to co-exist.
- Fedora may be considered a more community happy one like Ubuntu. Most dependencies could be easily resolved like Ubuntu. (Need to check later)
- howdy for CentOS gotta be the same as my POC work on RHEL.
- I will look into both CentOS and Fedora (over the weekend)
- Packaging the main howdy is the least of our worries? 

> STEP 0 (Base OS : RHEL7.5)
> STEP 1 (Python 3.x.x)
> STEP 2 (Prep Prep Prep for face_recognition and dlib)
> STEP 3 (face_recognition (with dlib))
> STEP 4 (Another Prep Prep Prep for fswebcam and making it)
> STEP 5 (Prep Prep Prep for pam_python.so) 
> STEP 6 (Final Touch)
> STEP 7 (Finally, Behold! howdy!)
> STEP 8 (Good Luck! Test Test Test)

> STEP 0 (Base OS : RHEL7.5)

RHEL : 7.5 (Server with GUI) 
nano is there :)

> STEP 1 (Python 3.x.x)

sudo yum install openssl-devel bzip2-devel
wget https://www.python.org/ftp/python/3.6.7/Python-3.6.7.tar.xz
./configure --enable-optimizations
sudo make altinstall # Needs (2GB+ of) RAM and SWAP, Otherwise, it throws out funny errors.

> STEP 2 (Prep Prep Prep for face_recognition and dlib)

sudo vi /etc/sudoers # To add/append /usr/local/bin to Defaults secure_path so that sudo can find the new locally installed binaries easily.	
sudo pip3.6 install numpy
sudo pip3.6 install cmake
sudo pip3.6 install opencv-python

> STEP 3 (face_recognition (with dlib))
sudo pip3.6 install face_recognition # This will do dlib automatically. # Be very patient.

> STEP 4 (Another Prep Prep Prep for fswebcam and making it)
sudo yum install libjpeg-turbo-devel.x86_64
sudo yum install libpng
sudo yum install libpng-devel.x86_64
sudo yum install freetype-devel.x86_64
wget https://github.com/libgd/libgd/releases/download/gd-2.2.5/libgd-2.2.5.tar.gz
./configure
sudo make install

sudo ldconfig # After adding /usr/local/lib in /etc/ld.so.conf.d/libgd.conf
fswebcam hi.jpeg # Just to make sure it works.

> STEP 5 (Prep Prep Prep for pam_python.so) 

sudo pip3.6 install Sphinx
sudo yum install pam-devel.x86_64
sudo yum install python-devel.x86_64

wget -O pam-python_1.0.6.orig.tar.gz https://sourceforge.net/projects/pam-python/files/latest/download
vi pam_python.c # Workaround for error: #define _XOPEN_SOURCE 600 #move the #include for Python.h above the #include for pam_module
make
sudo make install
sudo cp /lib/security/pam_python.so /lib64/security/

> STEP 6 (Final Touch)

Let's add them on top of pam config files. common-auth of ubuntu is partly in system-auth of Red Hat. And gdm-password.
-----
# Howdy IR face recognition
auth	    sufficient	  pam_python.so /lib/security/howdy/pam.py
-----

sudo vi /etc/pam.d/system-auth # For sudo and stuff
sudo vi /etc/pam.d/gdm-password # For Gnome GUI login and stuff

sudo ln -s /usr/local/bin/python3.6 /usr/local/bin/python3
sudo ln -s /lib/security/howdy/cli.py /usr/local/bin/howdy 

> STEP 7 (Finally, Behold! howdy!)

git clone https://github.com/boltgolt/howdy.git
sudo mkdir /lib/security
sudo cp -Rf src/ /lib/security/howdy/
sudo cp autocomplete/howdy /usr/share/bash-completion/completions
sudo sed -i "s/usr/usr\/local/g" /lib/security/howdy/pam.py

udevadm info -r --query=all -n /dev/video0 | grep DEVLINKS # Get a web cam dev by-path like /dev/v4l/by-path/pci-0000:00:05.7-usb-0:1:1.0-video-index0
sudo howdy config # Configure device_path accordingly

> STEP 8 (Good Luck! Test Test Test)

It works for me :)

williamwlk, 2018-11-28:
Hi @boltgolt 

> Nice work! As Fedora and Redhat still use the same packaging process, would that mean that your work would also run on Fedora? 

It might. Just that I am concerned about how the dependencies are handled. I really need to go take a look on Fedora and CentOS. Been a while I have been away from those two.

> Would there be a big difference in installation instructions between CentOS, Fedora and RHEL?

CentOS and RHEL are version compatible and binary compatible. So I don't worry much really.

For Fedora, there is always a new version every six months (vs the life-cycle of CentOS/RHEL is min 2 years).  I really need to take a look. As I have mentioned, the dynamism of Fedora is very much like that Ubuntu in terms of community and updates. As such, we don't need to worry much too.

Too positive a thinking?

boltgolt, 2018-11-29:
Great write up, but those dependency installations truly look like absolute hell. 

If the Howdy installation itself is so relatively easy, would it be an idea to link Redhat users to a separate wiki page with instructions on how to install all dependencies before running the simple fedora copr installation?

williamwlk, 2018-11-30:
> Great write up, but those dependency installations truly look like absolute hell.

Thank God it is all Linux after all. Trial and Error based on a perfectly running Ubuntu Machine.

> 
> If the Howdy installation itself is so relatively easy, would it be an idea to link Redhat users to a separate wiki page with instructions on how to install all dependencies before running the simple fedora copr installation?

An OK idea to begin with under the assumption that we grab the main howdy rpm only (howdy-2.4.0-1.fc29.x86_64.rpm). My main worry is that the dependencies **_could_** break. I have not tried it on CentOS7 or RHEL7.5 yet but hopefully tonight I will.

But a more recommended approach is to host the stuff at EPEL for RHEL/CentOS as howdy-2.4.0-1.el7.x86_64.rpm along with other compatible dependencies?

williamwlk, 2018-11-30:
Hi @boltgolt Hi @luyatshimbalanga 

It breaks within the CentOS eco system. :(

> CentOS Linux release 7.5.1804 (Core) 

Error: Package: python3-dlib-19.16-1.fc29.x86_64 (luya-howdy)
           Requires: libstdc++.so.6(GLIBCXX_3.4.20)(64bit)

Error: Package: python3-dlib-19.16-1.fc29.x86_64 (luya-howdy)
           Requires: libstdc++.so.6(CXXABI_1.3.8)(64bit)

[w@mycentos0 ~]$ strings /lib64/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_DEBUG_MESSAGE_LENGTH
[w@mycentos0 ~]$ strings /lib64/libstdc++.so.6 | grep CXXABI_1.3
CXXABI_1.3
CXXABI_1.3.1
CXXABI_1.3.2
CXXABI_1.3.3
CXXABI_1.3.4
CXXABI_1.3.5
CXXABI_1.3.6
CXXABI_1.3.7

Of course, I can fix it  by simply a copy/paste of that lib file among other things, but it is not the way it is. 

I think we should maintain the eco system differently. What do you guys think?

Thoughts?

luyatshimbalanga, 2018-12-01:
Hi @williamwlk ,
You will need to install [Fedora EPEL repository](https://fedoraproject.org/wiki/EPEL) on CentOS. If you can list the dependencies from CentOS/RHEL for howdy, I can update the spec file for building package for them.

williamwlk, 2018-12-02:
Hi @luyatshimbalanga 

EPEL is there and I have no issues installing the python3 stuff, which is one of the dependencies.

The issue is that you compiled python3-dlib with Fedora's system lib 'ibstdc++.so.6' with  (**_GLIBCXX_3.4.20_**)(64bit) whereby that of CentOS has just **_GLIBCXX_3.4.19_**. And likewise for CXXABI.

We/You will have to compile it on a designated CentOS System with an updated spec file for building the package.

Hope this helps.

Will

williamwlk, 2018-12-02:
@boltgolt @luyatshimbalanga 

I hope to give you my recommendations for RHEL/CentOS Distros tomorrow.

I have played around a lot already on my own and with the Copr repo of luya.

Time to wrap up :)

Cheers.

boltgolt, 2018-12-09:
Any progress @williamwlk?

williamwlk, 2018-12-09:
@boltgolt @luyatshimbalanga here we go :)

https://github.com/williamwlk/my-red-corner

Thoughts? Sorry took a while to get back to you guys on this.

Will work on RHEL7. Won't be an issue.

williamwlk, 2018-12-10:
@boltgolt @luyatshimbalanga 

My recommendation (for now) for the CentOS (or RHEL for that matter) is to go and package howdy alone without any 3rd parties packages (such as dlib, face_recognition, etc) and I have done a POC in that manner.

One of the reasons is that I do not have enough time to build the dependencies rpms. And those dependecies can be managed easily as stated in the following howto or manual.

> Sounds good to you both?

# You know the drill. Be root or a sudo user.

# We will need this.
yum groups install "Development Tools"

# We will need this. 
yum install https://dl.fedoraproject.org/pub/epel/epel-release-latest-7.noarch.rpm
# We need both python2(default) and python3(adding now).
yum install python36 python36-devel python36-setuptools

# Let pip do the walking. Security a concern? 
easy_install-3.6 pip
pip3 install cmake
# opencv-python (numpy)
pip3 install opencv-python 
# face_recognition (face-recognition-models, dlib)
pip3 install face_recognition

# A good webcam utility for fun (jkd) for later use or for testing purposes
# A selfie by #fswebcam smile.jpeg
yum install http://rpms.remirepo.net/enterprise/7/remi/x86_64//gd-last-2.2.5-5.el7.remi.x86_64.rpm
yum install https://rpmfind.net/linux/fedora/linux/development/rawhide/Everything/x86_64/os/Packages/f/fswebcam-20140113-10.fc29.x86_64.rpm

# pam_python
yum install http://widehat.opensuse.org/opensuse/repositories/home:/zhonghuaren/Fedora_28/x86_64/pam_python-1.0.6-2.1.x86_64.rpm

# This is it.
yum install https://github.com/williamwlk/my-red-corner/raw/master/howdy-2.4.0-2.el7.x86_64.rpm

# Sigh
ln -s /usr/bin/python3.6 /usr/bin/python3

# Sigh
echo "/usr/local/lib/python3.6/site-packages" > /usr/lib/python3.6/site-packages/usrlocal.pth
echo "/usr/local/lib64/python3.6/site-packages" > /usr/lib64/python3.6/site-packages/usrlocal.pth

# Let me configure howdy for you.
wget https://raw.githubusercontent.com/williamwlk/my-red-corner/master/let_me_configure_howdy_for_you.sh && chmod +x let_me_configure_howdy_for_you.sh
./let_me_configure_howdy_for_you.sh

# That's it. I guess.
# ¯\_(ツ)_/¯"

boltgolt, 2018-12-10:
I'll add it as a wiki page on here and link to it from the readme. Thanks a lot!

closed 2018-12-10 by boltgolt