boltgolt/howdy #104: howly test not running well on Wayland session

luyatshimbalanga, 2018-11-29:
_Please describe the issue in as much detail as possible, including any errors and traces_

An issue occurs when running **howdy** in a Wayland session like Gnome Shell.
```
 sudo howdy test

[gPhoto2][W]: opening device failed: gp_list_get_name(allDevices, index, &model): Bad parameters

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
No protocol specified
qt.qpa.screen: QXcbConnection: Could not connect to display :0
Could not connect to any X display.
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):
Fedora 

Howdy version: 2.4.0

luyatshimbalanga, 2018-11-30:
```
sudo QT_QPA_PLATFORM=wayland howdy test
[gPhoto2][W]: opening device failed: gp_list_get_name(allDevices, index, &model): Bad parameters
VIDEOIO ERROR: V4L: device none: Unable to query number of channels
GStreamer: Error opening bin: no element "none"
CvCapture_OpenNI::CvCapture_OpenNI : Failed to open input file (none): Bad file extension

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
error: XDG_RUNTIME_DIR not set in the environment.
Failed to create wl_display (No such file or directory)
qt.qpa.plugin: Could not load the Qt platform plugin "wayland" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

Aborted
```
Same result despite reinstallation.

sapjunior, 2018-12-01:
Try this, https://www.reddit.com/r/Fedora/comments/5eb633/solution_running_graphical_app_with_sudo_in/. The problem comes from sudo which is tied to X.org

luyatshimbalanga, 2018-12-01:
Thanks for the link. Interestingly, a bug report with mentioned workaround the heating debate related to using root on graphical application. Perhaps it would be worth to take a look at [polkit](https://www.freedesktop.org/software/polkit/docs/latest/)as a better authentication method because of fine control. [Here is an example](https://wiki.archlinux.org/index.php/Polkit).

boltgolt, 2018-12-02:
Closing this as it's not directly related to Howdy. Thanks a lot for finding a solution @sapjunior!

closed 2018-12-02 by boltgolt