boltgolt/howdy #107: Use pam-auth-update for pam inclusion

boltgolt, 2018-12-05:
Right now, Howdy is inserting a line in `/etc/pam.d/common-auth` in de debian installer. This is a dangerous file to edit and that's why there is a confirmation step. This step is rather hostile to users not familiar with PAM modules though.

By making use of the `pam-auth-update` command Howdy could be included into PAM at no risk, and be easily removed when the package is uninstalled.

williamwlk, 2018-12-14:
Hi @boltgolt 

CentOS/RedHat (and Fedora) Platform too. I have to deal with PAM. Not pleasant but no choice.

FYI:
https://github.com/williamwlk/my-red-corner/blob/master/README_PAM.txt

boltgolt, 2018-12-14:
Do those distros have `pam-auth-update`?

williamwlk, 2018-12-14:
Nope. 

But we have got authconfig

https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/system-level_authentication_guide/authconfig-install

williamwlk, 2018-12-14:
And we don't have /etc/pam.d/common-auth or common-* for that matter.

But we have got them all apparently combined into this :

/etc/pam.d/system-auth (system-auth-ac)

Hope this helps.

williamwlk, 2018-12-15:
We have got /etc/pam.d/gdm-fingerprint and it is an interesting approach. I should pay more attention to this risk-free scenario where we can safely add or create a pam configuration file for howdy.

I will look into this and give you my thoughts about it based on CentOS/RHEL design.

Buy me some time though. Thank you.

boltgolt, 2018-12-17:
I can't imagine there isn't a save and automatic way to be included in PAM on those distros, that's key for so many uses. However, even the Yubikey setup requires the user to manually add a line to `/etc/pam.d/login` so maybe not.