boltgolt/howdy #112: Thinkpad x1 extreme IR camera issue

ericvhileman, 2018-12-19:
_Please describe the issue in as much detail as possible, including any errors and traces_

Authentication fails using IR camera on Thinkpad x1 Extreme.  The first problem is that the test image shows snow and the IR emitters don't light up at all.  Editing the config to force_mjpeg true shows the grayscale image, the IR emitters light up, and allows a profile to be added.  However, the profile won't auth unless I slightly block either IR emitter from the edge.  For example if I hold out my left thumb and move it in about 1" away from the side of left IR emitter I will auth.  I can also confirm this on the test image by moving any object in to kind of block either IR emitter from the outer edge and the red circle appears. Slightly blocking from either side works.  Using my regular camera works fine.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Kubuntu 18.10 x11 plasma.

Howdy version: latest

boltgolt, 2018-12-20:
This might be a dark frame detection issue, could you run `sudo howdy test` and look at the right left of the frame? Depending on your type of IR emitters is should rapidly switch between "dark frame" and "scan frame". Dark frames will not be scanned for faces and will fail.

If this is the issue you could try playing with the `dark_threshold` config value.

ericvhileman, 2018-12-20:
Definitely some sort of dark issue but I think it's the whole picture and not just a dark frame.  In the test window top right it's always dark frame.  I see now what moving my thumb into the IR emitter does is light up image more.  I'm sitting close with my laptop on my desk and leaning forward.  My face and body are in the image fine with nothing but blackness around me.  Moving my thumb into the IR emitter lights up more around me and I get a match.  It also works if I lean way back and I can see the captured image also shows more around me.  So when it's just my face and body there's not match "dark frame".  When my image has my background around me it matches.

I also tried tweaking the dark frames config, 1/25/50/75/100/1000, it had not effect. Sounds like that's more for frames and not the image itself.

d3sandoval, 2018-12-20:
Here's what the howdy test looks like on my machine (i have also played around with the dark frames config):
[image]

d3sandoval, 2018-12-20:
Editing the config to force_mjpeg true seems to fix the issue.

ericvhileman, 2018-12-20:
@d3sandoval I had that same problem initially and force_mjpeg got my IR camera working.  But now I'm stuck with the dark frame issue.  Do you have that issue as described by me?

d3sandoval, 2018-12-20:
I will update this thread if the dark frame issue comes back. So far it seems to unlock reliably after registering a few new models

ericvhileman, 2018-12-20:
@d3sandoval are you sitting close to your laptop webcam (within 1-2 feet) or leaning back away (2-5 feet)?

d3sandoval, 2018-12-20:
In the screen shot I pasted, those green frames appear no matter how far away i am from the camera. Using force_mjpeg, i was able to create models with my face filling the entire frame, one of me 1-2ft and one about 4 feet back. Howdy was able to detect me each time.

ericvhileman, 2018-12-20:
Here's some example screenshots.  This first one is a fail when I'm close and that looks like it's too much darkness:
[image]

But if I lean back, the image brightens up by showing my background, and it's a success:
[image]

None of the dark frame settings appear to have any effect...

ericvhileman, 2018-12-20:
I commented out line 116 in compare.py so it doesn't continue and now it's working.

boltgolt, 2018-12-20:
Setting `dark_threshold` to 100 should have the same effect, but i'm glad it's working now!

closed 2018-12-20 by boltgolt

ericvhileman, 2018-12-21:
Updating this issue... I put some debug statements in and see that at times "float(hist[0]) / hist_total * 100" was 99.9999 so setting to 100 works.  No need to comment out the continue.  I think the reason why it didn't work before was I only had one profile captures.  Once I captured a few more and set it to 100 it works reliably.  Cool!