boltgolt/howdy #119: TypeError: compute_face_descriptor()

Marteon, 2019-01-06:
Device: Dell XPS 13 infrared camera

howdy version: '40c290e' (Version: 2.5.0)

```
Traceback (most recent call last):
  File "/usr/bin/howdy", line 90, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 117, in <module>
    enc = face_recognition.face_encodings(frame)
  File "/usr/lib/python3.7/site-packages/face_recognition/api.py", line 210, in face_encodings
    return [np.array(face_encoder.compute_face_descriptor(face_image, raw_landmark_set, num_jitters)) for raw_landmark_set in raw_landmarks]
  File "/usr/lib/python3.7/site-packages/face_recognition/api.py", line 210, in <listcomp>
    return [np.array(face_encoder.compute_face_descriptor(face_image, raw_landmark_set, num_jitters)) for raw_landmark_set in raw_landmarks]
TypeError: compute_face_descriptor(): incompatible function arguments. The following argument types are supported:
    1. (self: dlib.face_recognition_model_v1, img: numpy.ndarray[(rows,cols,3),uint8], face: dlib.full_object_detection, num_jitters: int=0, padding: float=0.25) -> dlib.vector
    2. (self: dlib.face_recognition_model_v1, img: numpy.ndarray[(rows,cols,3),uint8], faces: dlib.full_object_detections, num_jitters: int=0, padding: float=0.25) -> dlib.vectors
    3. (self: dlib.face_recognition_model_v1, batch_img: List[numpy.ndarray[(rows,cols,3),uint8]], batch_faces: List[dlib.full_object_detections], num_jitters: int=0, padding: float=0.25) -> dlib.vectorss

Invoked with: <dlib.face_recognition_model_v1 object at 0x7f9e4d7fd0a0>, array([[ 51,  49,  50, ...,  52,  49,  51],
       [ 49,  49,  48, ...,  47,  50,  49],
       [ 50,  50,  51, ...,  49,  49,  50],
       ...,
       [ 97,  96,  95, ..., 191, 184, 189],
       [ 96,  98,  97, ..., 195, 193, 188],
       [ 94,  97,  97, ..., 198, 194, 192]], dtype=uint8), <dlib.full_object_detection object at 0x7f9e4d7fd308>, 1
```

boltgolt, 2019-01-06:
This exact error has already been reported at #70. Try switching to FFmpeg as recorder in de config.

closed 2019-01-06 by boltgolt

c02kr, 2023-04-07:
hey @ghost have you figured it out?? 
please help if you have.
thanks

c02kr, 2023-04-07:
I can't seem to understand

javifriasj, 2023-04-07:
Hi, 

I am having the same issue.. have someone solve it?

Thanks.

Halibayyb, 2024-05-05:
Hello guys, I found the solution and it worked you can find it from this GitHub link: https://stackoverflow.com/questions/75926662/face-recognition-problem-with-face-encodings-function

I replaced 
`rgb_small_frame = cv2.cvtColor(small_frame, cv2.COLOR_BGR2RGB)` with `rgb_small_frame = frame_process[:, :, ::-1]`