boltgolt/howdy #120: Auth for installing apps in ubuntu not working?

timwelch, 2019-01-08:
For some reason, when I run anything in the GUI in ubuntu (i.e. the app installer, or the kernel updater), it attempts to auth and my IR cameras come on, but it just loops forever and ever and eventually times out and asks for a password. All other auths on the system (sudo, password at login prompt, etc) are working as desired so I'm not sure what the issue is.

I have added the pam line to gdm-password and gdm-launch-environment. Although, that it begins the auth process by turning on the IR camera tells me it's already working from common-auth... 

The error i get in /var/log/auth.log is this:

```
Jan  8 16:47:29 loveprayserve polkitd(authority=local): Operator of unix-session:4 FAILED to authenticate to gain authorization for action io.snapcraft.snapd.manage for unix-process:2820:7897 [/usr/bin/gnome-software --gapplication-service] (owned by unix-user:tim)

```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Ubuntu 18.10

Howdy version:

Latest from Git, just before the 2.5.0 package was dropped. :-)

dmig, 2019-01-09:
Did you enable `end_report` in config?

timwelch, 2019-01-09:
Wow, yes it was. I didn't know that would make the GUI auths fail. 
Thanks for the help!

closed 2019-01-09 by timwelch

dmig, 2019-01-09:
Actually, there is a warning in config about this issue, but for some reason comments get wiped :(

dmig, 2019-01-09:
The reason is: `pkexec` doesn't like output from auth plugin. Maybe directing it to `stderr` will fix this issue.

boltgolt, 2019-01-09:
The comments are removed by the config migration between versions. When importing the new config file the library ignores all comments, so they can't be written to file at the end of the installation either. I'll look into fixing that in the next release.

That's a weird issue, thanks a lot to @dmig for figuring it out! As `end_report` is a debug toggle i don't think it's warranted to move its output to stderr (especially because it's not an error), but i'll add a warning to the debug config section as a whole.