boltgolt/howdy #121: Face Unlock not working after explicit locking the screen in Linux Mint

itsgauravjain22, 2019-01-10:
_Please describe the issue in as much detail as possible, including any errors and traces_

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version:

boltgolt, 2019-01-10:
Could your run `sudo howdy config` and set no_confirmation to `true`? Also check that end_report has been set to `false`.

Are there any errors in `/var/log/auth.log`?

itsgauravjain22, 2019-01-11:
I am unaware of howdy config. Thanks for the help. It solved the problem of not login after the explicit log off the laptop. But tthe above mentioned comment is a bug with set `no_confirmation` to `true`.

I have uploaded below my `/var/log/auth.log` for further information
[gaurav-auth.log](https://github.com/boltgolt/howdy/files/2748464/gaurav-auth.log)

boltgolt, 2019-01-11:
Unfortunately this is a problem with the way the screensavers handle PAM conversations, good to hear the fix worked though!

closed 2019-01-11 by boltgolt

haakon8855, 2019-12-09:
How did you solve this?

sethjt33, 2021-01-14:
Hello, at first I was not able to get howdy to start on the lock screen however, after making these changes it starts now and the camera comes on but it fails every time to unlock from the lock screen.  All other application of howdy works perfectly with exception of the lock screen.  does anyone have any ideas?

chukynax, 2022-03-23:
same thing here, ubuntu xfce4 desktop, light-locker 
Mar 23 09:33:52 dell /lib/security/howdy/pam.py[10210]: Traceback (most recent call last):
Mar 23 09:33:52 dell /lib/security/howdy/pam.py[10210]:   File "/lib/security/howdy/pam.py", line 93, in pam_sm_authenticate
Mar 23 09:33:52 dell /lib/security/howdy/pam.py[10210]:     return doAuth(pamh)
Mar 23 09:33:52 dell /lib/security/howdy/pam.py[10210]:   File "/lib/security/howdy/pam.py", line 39, in doAuth
Mar 23 09:33:52 dell /lib/security/howdy/pam.py[10210]:     pamh.conversation(pamh.Message(pamh.PAM_TEXT_INFO, "Attempting face detection"))
Mar 23 09:33:52 dell /lib/security/howdy/pam.py[10210]: PamException: Conversation error

jkatnik, 2023-03-02:
Hi @boltgolt in 2023 :)

is there any progress regarding unlocking Mint (21.1) from the screensaver?
It works in Ubuntu 22, so maybe there is a way to make it run in Mint too?

egavazzi, 2023-09-16:
@jkatnik @SeidlEnt [This comment](https://github.com/boltgolt/howdy/issues/28#issuecomment-396231042) worked for me on Mint 21.2 :)