boltgolt/howdy #122: Update AUR (Possible addition of -git package?)

joeknock90, 2019-01-11:
The AUR PKGBUILD is out of date for 2.5.0

I've attached a working PKGBUILD for 2.5.0 if you want to use it. 

Might I also suggest a howdy-git package to build from github directly from the AUR?

Thanks!

[PKGBUILD-2.5.0.txt](https://github.com/boltgolt/howdy/files/2749768/PKGBUILD-2.5.0.txt)

boltgolt, 2019-01-12:
Thanks for providing your modified PKGBUILD! Some major things have changed in Howdy between 2.3.1 and 2.5.0, notably including removing face-recognition as an dependency and requiring extra optimization steps for dlib compilation. I currently don't see these changes in the PKGBUILD and i'm hesitant to push a working but incomplete script at the moment.

boltgolt, 2019-01-12:
If you want to, you could port [the changes](https://github.com/boltgolt/howdy/compare/v2.3.1...master#diff-07e2cd25f7da0550504d107fb2cae5be) (press "Files changed") from the debian installer to the PKGBUILD. The wiki is also out of date for 2.5.0 i think.

boltgolt, 2019-01-16:
Let me know if you have any issues figuring out what the postinst is doing

depadiernos, 2019-02-24:
Just wondering if this postinst python script can just be use as-is as the post_install function mentioned here: https://git.archlinux.org/pacman.git/plain/proto/proto.install. :thinking:

boltgolt, 2019-02-27:
Sounds good, and might also be able to automate the [uninstallation steps](https://github.com/boltgolt/howdy/blob/master/debian/prerm).

depadiernos, 2019-03-08:
Looks like somebody updated the AUR pkgbuild, but it's not working. :thinking: Getting:
```
127 dennis@zelda2 /usr/lib/security % sudo howdy add                          :(
Data files have not been downloaded, please run the following commands:

	cd /usr/lib/security/howdy/dlib-data
	sudo ./install.sh
```
But there is no install.sh in that folder.

boltgolt, 2019-03-08:
That folder should include a [readme](https://github.com/boltgolt/howdy/tree/master/src/dlib-data) as well with commands to run manually. It's something the something the PKGBUILD should do automatically in my opinion.

depadiernos, 2019-03-08:
Ah, I see!

pthalamy, 2019-03-20:
I am running into new issues updating howdy from AUR in its current state. I was previously running howdy 2.3.1 without issue.

The installation fails during the `package()` phase.

```
==> Entering fakeroot environment...
==> Starting package()...
Collecting face_recognition
  Using cached https://files.pythonhosted.org/packages/3f/ed/ad9a28042f373d4633fc8b49109b623597d6f193d3bbbef7780a5ee8eef2/face_recognition-1.2.3-py2.py3-none-any.whl
Installing collected packages: face-recognition
  The scripts face_detection and face_recognition are installed in '/home/pthalamy/.cache/yay/howdy/pkg/howdy/usr/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed face-recognition-1.2.3
The --yes options to dlib's setup.py don't do anything since all these options 
are on by default.  So --yes has been removed.  Do not give it to setup.py.
==> ERROR: A failure occurred in package().
    Aborting...
Error making: howdy
```

I believe it might be related to the reference to CUDA during the build of `dlib`, as it is not available on my system, which uses an Intel integrated GPU. This is what comes up during the CMake setup : 

```
-- Found BLAS and LAPACK via pkg-config
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) (Required is at least version "7.5")
-- DID NOT FIND CUDA
-- Disabling CUDA support for dlib.  DLIB WILL NOT USE CUDA
```

I can post this issue as a standalone issue if you believe it is not directly related to this thread and the state of the AUR release. 

Linux distribution: `Linux 5.0.1-1-MANJARO` 
Howdy version: `2.5.0-1`

boltgolt, 2019-03-20:
It's because the `--yes` flag has been deprecated, but is still used in the PKGBUILD. Could you try removing it before installing?

pthalamy, 2019-03-20:
Running the `setup.py` with `python3 setup.py install --no DLIB_USE_CUDA --root="$pkgdir/" --optimize=1 --skip-build` worked indeed.

I then had to manually run the `install.sh` script as well of course.

boltgolt, 2019-03-26:
I'll remove the `--yes` flag in the next release.

depadiernos, 2019-05-07:
Seems like this PR addresses some of the concerns in this issue: httpss://github.com/boltgolt/howdy/pull/186. What do you think?

vanbreukelingen, 2019-06-19:
[miranda@archlaptop yaourt]$ sudo howdy add
Unknown error: -4
[sudo] password for miranda: 
Illegal instruction

is what I still get with an 

> Linux archlaptop 5.1.11-1-ck-silvermont #1 SMP PREEMPT Tue Jun 18 17:34:41 EDT 2019 x86_64 GNU/Linux

I'd to switch to dev branch?

OnlyCS, 2022-11-16:
#582 the -git has been created